rufengsuixing/luci-app-adguardhome #2: 安装1.7-1release版本无法启动

YujiaCheng1996, 2019-11-27:
从1.6升级后报错，尝试清除数据重装依旧报错，错误信息如下：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/general'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:38: Model 'AdGuardHome' not found!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

rufengsuixing, 2019-11-27:
试一下fix没
https://github.com/rufengsuixing/luci-app-adguardhome/releases/tag/1.7-2

YujiaCheng1996, 2019-11-27:
问题依旧，升级提示：
#opkg install luci-app-adguardhome_1.7-2_all.ipk 
Upgrading luci-app-adguardhome on root from 1.6-3 to 1.7-2...
Terminated
Removing obsolete file /usr/lib/lua/luci/model/cbi/AdGuardHome.lua.
Removing obsolete file /usr/lib/lua/luci/model/cbi/AdGuardHomelog.lua.
Configuring luci-app-adguardhome.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/AdGuardHome is different from the conffile in the new package. The new conffile will be placed at /etc/config/AdGuardHome-opkg.
报错：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/general'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:38: Model 'AdGuardHome' not found!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

rufengsuixing, 2019-11-27:
相关行已经改掉了，你安装的应该不是新版
entry({"admin","services","AdGuardHome","base"},cbi("AdGuardHome/base"),_("Base Setting"),1).leaf = true
另请删除 /tmp/luci-indexcache 和  /tmp/luci-modulecache/*

rufengsuixing, 2019-11-27:
代码中已加入删除luci缓存，再试试，如果没问题了可以关闭issue

rufengsuixing, 2019-11-28:
又尝试修复了一次再测试一下

YujiaCheng1996, 2019-11-28:
使用最新版已经修复，谢谢

closed 2019-11-28 by YujiaCheng1996