rufengsuixing/luci-app-adguardhome #3: 打开管理界面报错

garypang13, 2019-11-27:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/general'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua:35: attempt to concatenate local 'httpport' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

garypang13, 2019-11-27:
> /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin/services/AdGuardHome'.
> The called action terminated with an exception: [...]

是还需要装https://github.com/happyzhang1995/luci-app-adguardhome这个主程序吗?

rufengsuixing, 2019-11-27:
如下相关行已经改掉了，你安装的应该不是新版
entry({"admin","services","AdGuardHome","base"},cbi("AdGuardHome/base"),_("Base Setting"),1).leaf = true
另请删除 /tmp/luci-indexcache 和 /tmp/luci-modulecache/*

garypang13, 2019-11-27:
> 不需要安装主程序如下相关行已经改掉了，你安装的应该不是新版
> entry({"admin","services","AdGuardHome","base"},cbi("AdGuardHome/base"),_("Base Setting"),1).leaf = true
> 另请删除 /tmp/luci-indexcache 和 /tmp/luci-modulecache/*

用最新源码重新编译了次,打开后台 admin/services/AdGuardHome还是报一摸一样的错误,

rufengsuixing, 2019-11-28:
又尝试修复了一次，如果不行的话，我估计需要别人帮忙了，请看最新release，

garypang13, 2019-11-28:
> 又尝试修复了一次，如果不行的话，我估计需要别人帮忙了，请看最新release，

还是不行,一摸一样的错误,不管是用最新release还是下载最新源码自己编译都是一样.

rufengsuixing, 2019-11-28:
试试最新的release

garypang13, 2019-11-28:
> 试试最新的release

完美运行,感谢!

closed 2019-11-28 by garypang13