rufengsuixing/luci-app-adguardhome #5: 重定向53端口无效

garypang13, 2019-11-28:
* **AdGuard Home luci的版本:**
1.7-7
* **AdGuard Home server的版本:**
v0.99.3
* **操作系统和版本:**
OpenWrt R9.11.23 / LuCI Master (git-19.331.11944-c1039b9)
--
重定向方式选择重定向53端口到AdGuardHome不起作用,只能选择作为dnsmasq的上游服务器,但是作为dnsmasq的上游服务器的话会导致在AdGuardHome中所有客户端都显示为了127.0.0.1,没办法再针对客户端做精确控制了,
不知道是不是和最新版的lean lede有关,没更新lede前53重定向还能正常使用,更新后就不行了,

rufengsuixing, 2019-11-28:
lean的自定义固件对53端口进行了重定向，而我的规则是添加在规则链的末尾，如果想要使用请清空防火墙的自定义规则，链接为 http://192.168.1.1/cgi-bin/luci/admin/network/firewall/custom

garypang13, 2019-11-29:
我现在退回到了OpenWrt R9.10.24,其他所有设置都没变,53端口重定向和SSR PLUS功能都完全正常,SSR Plus的解析方式设置的使用本机端口为5335的DNS服务,

rufengsuixing, 2019-11-29:
lean固件53端口重定向的优先级要比我的重定向要高，
应该是所有的dns请求都通过了dnsmasq，在dnsmasq中进行匹配，其中gfwlist域名通过了adguard，adguard显示的查询应该全来自路由器

rufengsuixing, 2019-11-29:
我提供给你的临时解决方案是，删掉自定义的53端口重定向，使用adguard5335的重定向，ssr-plus使用5335的dns服务，然后adguard使用原来的dnsmasq 53作为唯一上游服务

rufengsuixing, 2019-11-29:
我提供给你的临时解决方案是，删掉自定义的53端口重定向，使用adguard的重定向，ssr-plus使用pdnsd，然后adguard使用原来的dnsmasq 127.0.0.1:53作为唯一上游服务，删掉adguard里面的gfw列表，应该就可以了

rufengsuixing, 2019-11-29:
或者修改掉dnsmasq的端口，然后adguard使用53端口，指定dnsmasq为adguard唯一上游

garypang13, 2019-11-29:
防火墙自定义规则中确实有两条
iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53
我把第一条注释掉,adguardhome的53重定向就生效了,SSR PLUS需要设置为绕过中国大陆IP模式,不能用gfwlist模式,不知道会不会有其他什么副总用,
感谢!

rufengsuixing, 2019-11-29:
我提供给你的方案就是为了作gfw模式的，如果是china绕过模式，直接用adguardhome作为dns服务器就可以了，不用折腾

closed 2019-11-29 by rufengsuixing

Jieoz, 2021-07-14:
> 我提供给你的方案就是为了作gfw模式的，如果是china绕过模式，直接用adguardhome作为dns服务器就可以了，不用折腾

"指定dnsmasq为adguard唯一上游"这一步怎么做啊大佬，我也遇到这个问题，想用gfw模式，但是一直都连接超时。

我现在把dnsmasq改成5353，adg是53端口，ssr里设置的5335的dns服务。用绕过国内IP可以打开GoogleYouTube，gfw就显示超时

wangliangliang2, 2023-02-27:
这个issue的问题如下：
kiddin9/openwrt-adguardhome#2
解决办法就是 -A 改成-I

OvidioLee, 2024-02-07:
这两组 `iptables` 命令的主要区别在于使用的是 `-A`（追加）选项还是 `-I`（插入）选项。这两个选项都用于向规则链中添加新规则，但它们在添加规则的位置上有所不同：

- `-A`（Append）选项会将新规则追加到指定链的末尾。这意味着，如果有多条规则，新添加的规则会被评估在最后。如果前面的规则已经匹配并处理了流量，那么追加的规则将不会被触及。

- `-I`（Insert）选项将新规则插入到指定链的开头（或指定的位置）。这意味着新插入的规则会首先被评估。使用 `-I` 选项添加的规则会优先于之前存在的规则，这对于确保新规则有最高的匹配优先级非常重要。

在你提供的命令中，使用 `-I` 选项将UDP和TCP目的端口为53的流量重定向到本地的5353端口的规则被插入到 `PREROUTING` 链的开头。这确保了无论之前 `PREROUTING` 链中有什么规则，这些新的重定向规则都会首先被评估和执行。这在你需要确保DNS流量被重定向到指定端口，而不被任何先前定义的规则影响时非常有用。

总结来说，使用 `-I` 选项可以让你确保新规则有最高的处理优先级，特别是在需要覆盖或预先处理特定流量类型之前的场景中非常有用。