rufengsuixing/luci-app-adguardhome #6: 请问大佬,这个OPENWRT的adguard home 是否可以屏蔽一些视频网站的ad

wtjfree, 2019-11-29:
请问大佬,这个OPENWRT的adguard home 是否可以屏蔽一些国内视频网站的广告呢?

和PC版的功能上有多大区别知道吗?

rufengsuixing, 2019-11-29:
adguardhome 是基于host的 host能做的很有限，如果广告和视频是同一个域名下的，无法屏蔽，但是胜在对资源消耗比较少
pc端的不清楚，我只用过pc浏览器端的，pc浏览器端的就会更加给力一点，因为可以对内容进行审查，发现同域名的广告元素也会进行移除，相对的对于性能要求也会高很多，当然电脑和路由器性能不能相比。所以能用浏览器端的就没有路由器端什么事了。
这两者都需要的最重要的东西是规则，再好的过滤器没有规则也是没有用武之地，所以虽然已经有路由器端的各种内容过滤器比如adbyby，adm，koolproxy，但是现在规则已经不更新或者对于性能需求太高而被我舍弃。
另外adguard的规则稍稍有点水土不服，浏览器端尝试勾选chinalist，路由器端可以尝试yhost，自己找到一个可靠的规则更新可能比过滤器还重要。
所以路由器端主要是为了不会用浏览器端的插件或者不便于用浏览器插件的设备准备的，比如手机和电视。
总结：能用浏览器插件尽量用插件，不能用的上路由器。

wtjfree, 2019-11-29:
非常感谢.详尽的解答.和这样好的项目. 我也只是只言片语的了解到国内视频网站,好多是HTTPS的加密广告.  在安卓上一般需要ROOT和加载一个根证书的方式来屏蔽广告, 

anyway还是要折腾起来.o(∩_∩)o 哈哈. 再次感谢.

closed 2019-11-29 by wtjfree

adaaaaaa, 2019-12-08:
> 相对的对于性能要求也会高很多
> 对于性能需求太高而被我舍弃。

我软路由的CPU是i3-9300T.。。。性能顶得住。。。请求加强功能。。。🤪🤪🤪

adaaaaaa, 2019-12-08:
> 请求加强过滤的能力。。。🤪🤪🤪

加强过滤能力仅仅增加列表就能完全实现吗？？

rufengsuixing, 2019-12-08:
内容审查过滤请使用koolproxy

garypang13, 2019-12-24:
@rufengsuixing @adaaaaaa 
adguardhome开发者确认未来会增加内容审查过滤的,
https://github.com/AdguardTeam/AdGuardHome/issues/677#issuecomment-477755152

rufengsuixing, 2019-12-24:
短时间是不会的，dns还没搞明白呢

Karmylr, 2020-12-27:
adguard home不能屏蔽https协议的广告 kp可以