rufengsuixing/luci-app-adguardhome #12: error

jkle112, 2019-12-02:
openwrt x64

menual config

<h2 name="content">500 Internal Server Error</h2>
<p>Sorry, the server encountered an unexpected error.</p>
<pre class="error500">/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/manual'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/ucisection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tvalue'.
A runtime error occured: /usr/lib/lua/nixio/fs.lua:25: bad argument #1 to 'open' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94></pre>

   <footer>
    <a href="https://github.com/openwrt/luci">Powered by LuCI Master (git-19.326.48450-5ffbb31)</a> / OpenWrt  R9.11.23
    
   </footer>
   </div>
  </div>
 </body>
</html>

rufengsuixing, 2019-12-02:
is there a config file in the path of configpath？

rufengsuixing, 2019-12-03:
you can test the lastest version, if not work reopen the issue

closed 2019-12-03 by rufengsuixing