rufengsuixing/luci-app-adguardhome #7: 问题反馈

fuqiang03, 2019-12-02:
[image]
[image]
openwrt master
必须手动shell 才可以运行

rufengsuixing, 2019-12-02:
没能复现，需要更多信息

fuqiang03, 2019-12-02:
openwt 主库代码 和luci  主库代码
控制页面无法启动adguardhome

rufengsuixing, 2019-12-02:
给更详细的信息，不然我也没办法，至少我这边没有复现

fuqiang03, 2019-12-02:
需要什么样的信息

rufengsuixing, 2019-12-02:
页面有没有报错或者提交时打开浏览器控制台看看返回的页面有没有502或者404

rufengsuixing, 2019-12-02:
或者系统日志报错信息

rufengsuixing, 2019-12-02:
可能是库调用问题，我可以尝试修复一下

fuqiang03, 2019-12-02:
root@OpenWrt:/# /etc/init.d/AdGuardHome start
fs error ln db to tmp /usr/bin/AdGuardHome overlay
这个是唯一的错误提示
只能用命令开启和关闭 
用命令开启后一切正常
可以进入页面

fuqiang03, 2019-12-02:
貌似只有新版的luci -modules出现这样的问题

rufengsuixing, 2019-12-02:
https://c-t.work/s/07b79766566a47
尝试一下

fuqiang03, 2019-12-02:
这回能启动了 但是enable 的值好像判断反了
[image]
这种状态是开启了

rufengsuixing, 2019-12-02:
没有bin文件的情况下会自动下载bin文件，需要时间

fuqiang03, 2019-12-02:
有bin的

fuqiang03, 2019-12-02:
希望发布个分支 然后我测试下

rufengsuixing, 2019-12-02:
我这边没有测试环境，我也不知道问题出在哪里，如果你有能力的话，可以试着提交pull request
相关代码在路由器的路径
/usr/lib/lua/luci/controller/AdGuardHome.lua 18-24行
\usr\lib\lua\luci\view\AdGuardHome\AdGuardHome_status.htm 
可以自行调试着修改，记得每次修改之后rm /etc/luci-indexcache 和rm /tmp/luci-modulecache/*
你可以测试一下如果bin路径为/usr/bin/AdGuardHomeAdGuardHome 时
pgrep /usr/bin/AdGuardHome/AdGuardHome  >/dev/null
echo $?
的返回值是否在进程存在时为0，不存在时为其他值

fuqiang03, 2019-12-02:
我的 我这头测试成功后 会做个patch 
然后贴在这里 没弄过pull request

rufengsuixing, 2019-12-02:
可以尝试删除各种local标签
比如
local uci=require"luci.model.uci".cursor()
改为
uci=require"luci.model.uci".cursor()

fuqiang03, 2019-12-02:
收到  我先Git一下 再次编译一下看看

rufengsuixing, 2019-12-02:
不用编译，直接在路由上修改就可以调试

fuqiang03, 2019-12-02:
好的

fuqiang03, 2019-12-02:
[image]

rufengsuixing, 2019-12-02:
这个结果没问题

fuqiang03, 2019-12-02:
[image]
改了他就正常了

fuqiang03, 2019-12-02:
[image]
enabled 的值 都乱套了

rufengsuixing, 2019-12-02:
[image]
和我这边的测试结果不同，我没法改

rufengsuixing, 2019-12-02:
是不是传参的问题把后面2>/dev/null的当成参数了

fuqiang03, 2019-12-02:
不会的
[image]
你的是版本是?

fuqiang03, 2019-12-02:
读取的配置文件enabled值 是正确的 但是执行的时候不知道什么干扰了这个值

rufengsuixing, 2019-12-02:
我这个是lean的op最新，
[image]
我可以在Pandora上再次测试
[image]

rufengsuixing, 2019-12-02:
就是说你的op版本
 test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
 test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
获得的值不同，其中前者是对的是么

fuqiang03, 2019-12-02:
后者是对的

rufengsuixing, 2019-12-02:
为啥，我枯了，这两个应该执行结果没区别，只是后者屏蔽了没有值的时候的报错信息

rufengsuixing, 2019-12-02:
另外我用的是config_get获取的enable信息，所以我觉得可能是op最新版的bug，先不予修复，op应该会自行修复这个bug的
config_get_bool enabled $CONFIGURATION enabled 0

rufengsuixing, 2019-12-02:
其他的应用enabled选项也反了么

fuqiang03, 2019-12-02:
没有

garypang13, 2019-12-02:
[image]

OpenWrt R9.11.23 / LuCI Master (git-19.331.11944-c1039b9)

fuqiang03, 2019-12-02:
lean的哦
知道了 
[image]
[image]

fuqiang03, 2019-12-02:
[image]
这都能反着执行

rufengsuixing, 2019-12-04:
> [image]
> 这都能反着执行
你能给出更多的测试结果么，关于uci get ，确保操作的过程中没有其他命令尝试改变开关状态

rufengsuixing, 2019-12-04:
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
echo $test
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
echo $test
尝试找一下其他配置文件里的flag量进行测试

closed 2019-12-05 by rufengsuixing