rufengsuixing/luci-app-adguardhome #8: luci开关

Mosney, 2019-12-02:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [Y] 我用着最新版的 luci-app-adguardhome
- [Y] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  * v0.99.3
* **AdGuard Home server的版本:**
  * v0.99.3
* **操作系统和版本:**
  * 
	OpenWrt 19.07.0-rc1 r10649-c4fdb377a2 / LuCI openwrt-19.07 branch git-19.332.23110-3e90684

> 勾选时，实际作用为禁用。不勾选时，实际作用为启用
> 
> [image]
>

Mosney, 2019-12-02:
手动编辑luci配置文件后解决，似乎是因为之前独立的ADG二进制在运行的缘故

Mosney, 2019-12-03:
启动脚本第9行开始似乎是把binpath和configpath写死了？
```
config_get configpath $CONFIGURATION configpath "/etc/AdGuardHome.yaml"
config_get binpath $CONFIGURATION binpath "/usr/bin/AdGuardHome/AdGuardHome"
```
实测会出现不按luci的配置放置/执行二进制的情况出现，另外同时也的确是存在开关反置现象，看了半天启动脚本和lua也没看出个所以然，只能描述一下观察到的现象。

Mosney, 2019-12-03:
贴一下目前的配置，在使用syslog时luci的日志显示功能也不正常。
```

config AdGuardHome 'AdGuardHome'
	option httpport '3000'
	option redirect 'none'
	option configpath '/etc/AdGuardHome.yaml'
	option verbose '0'
	option version 'v0.99.3'
	option gfwupstream ' '
	option workdir '/var/AdGuardHome/'
	option logfile 'syslog'
	option binpath '/usr/bin/AdGuardHome'
	option lastbinpath '/usr/bin/AdGuardHome'
	list old_redirect 'none'
	list old_port '53'
	list old_enabled '1'
	option enabled '1'

```

Mosney, 2019-12-03:
另外也不是很懂`gfw列表上游服务器`这个选项有什么用途，README里似乎也没有具体说明。

rufengsuixing, 2019-12-04:
开关相反是op新版luci的bug，在uci get 或者config get 时会获取和其他op版本相反的值，只涉及到最新版本op，不会修复，等待op官方修复

rufengsuixing, 2019-12-04:
syslog，确实写入日志是不能用的，这个是adh自己的bug

rufengsuixing, 2019-12-04:
gfw列表上游服务器在没有点击添加gfw加入列表之前是没有用的，如果用不到此功能，可以不用管这个值

rufengsuixing, 2019-12-04:
关于luci在脚本中获取配置的方式可以参照luci的wiki，你看到的写死是没有值时的默认值

rufengsuixing, 2019-12-04:
手动配置时，在配置提交前会调用二进制对配置文件进行检查，应该不会出现意外启动的情况

Mosney, 2019-12-04:
> syslog，确实写入日志貌似是不能用的，这个是adh自己的bug，另外能用的话也没有必要在这里显示系统日志

之前用二进制+配置文件跑的时候是可以写入syslog的，可能是通过`-l`指定的log文件不支持syslog？

rufengsuixing, 2019-12-04:
我测试了，syslog可用，并且我也更新了一下支持syslog日志在日志中显示，稍等我会发布release

rufengsuixing, 2019-12-04:
注意syslog有时候会失败，这是AdG的锅
/usr/bin/AdGuardHome/AdGuardHome -c /etc/AdGuardHome.yaml -
p 3000 -w /usr/bin/AdGuardHome -l syslog
2019/12/04 02:59:53 [fatal] cannot initialize syslog: Unix syslog delivery error

rufengsuixing, 2019-12-04:
你可以测试打印下
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
echo $test
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
echo $test

rufengsuixing, 2019-12-04:
https://github.com/rufengsuixing/luci-app-adguardhome/issues/7

rufengsuixing, 2019-12-04:
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
echo $test
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
echo $test
能提供下测试结果么，确保测试过程中没有修改enabled

liyiecho, 2019-12-04:
1对应停止运行，0对应运行中
[image]

[image]

liyiecho, 2019-12-04:
发现是`killall -q /usr/bin/AdGuardHome/AdGuardHome`这条命令的问题，执行这条命令无法结束程序运行

rufengsuixing, 2019-12-04:
如果path路径正确的话，这条命令应该没问题啊，可以去掉-q再试看看有没有报错

liyiecho, 2019-12-04:
killall: /usr/bin/AdGuardHome/AdGuardHome: no process killed
[image]
[image]

rufengsuixing, 2019-12-04:
这个确实是一个bug，我尝试修复一下

rufengsuixing, 2019-12-04:
你可以尝试release

liyiecho, 2019-12-04:
现在需要点两次保存并应用才能得到正确的，AdGuardHome.AdGuardHome.old_enabled的值对应相应的状态
[image]

rufengsuixing, 2019-12-04:
AdGuardHome.AdGuardHome.old_enabled 对应的只是对于重定向的设置，只影响到redirect，不会影响程序启动

liyiecho, 2019-12-04:
你看点两次保存并应用的结果里面enabled值不变，而old_enabled的值却改变了，按理来说，第二次点保存并应用不会改变old_enabled的值

rufengsuixing, 2019-12-04:
old 表示的是开关上一次的状态，连续两次应用，开关值不变，必然记录enable值

liyiecho, 2019-12-04:
enable值为0且old_enabled为1时，AdGuardHome还存活
[image]
enable值为0且old_enabled为0时，AdGuardHome停止运行
[image]

rufengsuixing, 2019-12-05:
总觉得是其他问题导致了表面上看到的开关相反的问题

closed 2019-12-05 by rufengsuixing

Mosney, 2019-12-05:
> test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
> echo $test
> test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
> echo $test
> 能提供下测试结果么，确保测试过程中没有修改enabled

1.7-8版本下正常使用状态下测得的结果
```
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
root@OP19:~# echo $test
1
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
root@OP19:~# echo $test
1
root@OP19:~# 
```
附上此时的配置 
```
root@OP19:~# uci show AdGuardHome
AdGuardHome.AdGuardHome=AdGuardHome
AdGuardHome.AdGuardHome.httpport='3000'
AdGuardHome.AdGuardHome.redirect='none'
AdGuardHome.AdGuardHome.configpath='/etc/AdGuardHome.yaml'
AdGuardHome.AdGuardHome.verbose='0'
AdGuardHome.AdGuardHome.version='v0.99.3'
AdGuardHome.AdGuardHome.workdir='/var/AdGuardHome/'
AdGuardHome.AdGuardHome.logfile='syslog'
AdGuardHome.AdGuardHome.binpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.lastbinpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.enabled='1'
AdGuardHome.AdGuardHome.old_redirect='none'
AdGuardHome.AdGuardHome.old_port='53'
AdGuardHome.AdGuardHome.old_enabled='1'
```

Mosney, 2019-12-05:
以下是1.7-16版本由开启状态取消勾选启用并保存应用后的测试结果 ，问题依旧
```
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
root@OP19:~# echo $test
0
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
root@OP19:~# echo $test
0
root@OP19:~# uci show AdGuardHome
AdGuardHome.AdGuardHome=AdGuardHome
AdGuardHome.AdGuardHome.httpport='3000'
AdGuardHome.AdGuardHome.redirect='none'
AdGuardHome.AdGuardHome.configpath='/etc/AdGuardHome.yaml'
AdGuardHome.AdGuardHome.verbose='0'
AdGuardHome.AdGuardHome.version='v0.99.3'
AdGuardHome.AdGuardHome.workdir='/var/AdGuardHome/'
AdGuardHome.AdGuardHome.logfile='syslog'
AdGuardHome.AdGuardHome.binpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.lastbinpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.old_redirect='none'
AdGuardHome.AdGuardHome.old_port='53'
AdGuardHome.AdGuardHome.old_enabled='1'
AdGuardHome.AdGuardHome.enabled='0'
AdGuardHome.AdGuardHome.gfwupstream='tcp://208.67.220.220#5353'

```
[image]

Mosney, 2019-12-05:
以下是1.7-16版本由上步实际运行状态勾选启用并保存应用后的测试结果 ，

```

root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
root@OP19:~# echo $test
1
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
root@OP19:~# echo $test
1
root@OP19:~# uci show AdGuardHome
AdGuardHome.AdGuardHome=AdGuardHome
AdGuardHome.AdGuardHome.httpport='3000'
AdGuardHome.AdGuardHome.redirect='none'
AdGuardHome.AdGuardHome.configpath='/etc/AdGuardHome.yaml'
AdGuardHome.AdGuardHome.verbose='0'
AdGuardHome.AdGuardHome.version='v0.99.3'
AdGuardHome.AdGuardHome.workdir='/var/AdGuardHome/'
AdGuardHome.AdGuardHome.logfile='syslog'
AdGuardHome.AdGuardHome.binpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.lastbinpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.gfwupstream='tcp://208.67.220.220#5353'
AdGuardHome.AdGuardHome.old_redirect='none'
AdGuardHome.AdGuardHome.old_port='53'
AdGuardHome.AdGuardHome.old_enabled='0'
AdGuardHome.AdGuardHome.enabled='1'

```

Mosney, 2019-12-05:
刚刚在上步状态下未作任何更改，仅重新点击了一下“保存并应用”按钮，提示“没有待应用的更改”，返回后发现状态变为运行，至此实际运行状态与开关匹配，配置文件数据如下所示。
```
root@OP19:~# uci show AdGuardHome
AdGuardHome.AdGuardHome=AdGuardHome
AdGuardHome.AdGuardHome.httpport='3000'
AdGuardHome.AdGuardHome.redirect='none'
AdGuardHome.AdGuardHome.configpath='/etc/AdGuardHome.yaml'
AdGuardHome.AdGuardHome.verbose='0'
AdGuardHome.AdGuardHome.version='v0.99.3'
AdGuardHome.AdGuardHome.workdir='/var/AdGuardHome/'
AdGuardHome.AdGuardHome.logfile='syslog'
AdGuardHome.AdGuardHome.binpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.lastbinpath='/usr/bin/AdGuardHome'
AdGuardHome.AdGuardHome.gfwupstream='tcp://208.67.220.220#5353'
AdGuardHome.AdGuardHome.enabled='1'
AdGuardHome.AdGuardHome.old_redirect='none'
AdGuardHome.AdGuardHome.old_port='53'
AdGuardHome.AdGuardHome.old_enabled='1'

root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
root@OP19:~# echo $test
1
root@OP19:~# test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
root@OP19:~# echo $test
1

```

Mosney, 2019-12-05:
根据以上三条，看起来是实际运行状态与`old_enabled`匹配。
状态与开关匹配之后与之前的配置文件就只有这一项存在差异，且可以明显看出在这个过程中发生变更的配置只有三条`old_`。
另外没看懂保留上次的部分配置有何意义。

Mosney, 2019-12-05:
https://github.com/rufengsuixing/luci-app-adguardhome/issues/8#issuecomment-562254554
经过重复测试，以上步骤可以稳定复现，基本可以确定是跟这三个上次状态的配置项有关系，实际问题应该是启动脚本逻辑判断上顺序的问题。
目前lean的luci相关代码均落后于目前官方分支的两个19.07的rc版本，应该是由于新版的luci导致这个问题暴露出来。

rufengsuixing, 2019-12-05:
结果如果和你说的一致，我的结论是，配置未完全写入时触发了reload，我会在reload前添加一个一秒的延时

liyiecho, 2019-12-06:
1.7-17版本，无论点几次“保存并应用”按钮，显示停止的还是停止不会启动，运行的还是运行不会停止
[image]
也不会改变old_enabled的值，old_enabled为1时，AdGuardHome还存活，只能手动`/etc/init.d/AdGuardHome stop`来停止
[image]
[image]
手动执行命令来看enable值没变，改变的是old_enabled，且old_enabled的值刚好对应状态
[image]

rufengsuixing, 2019-12-06:
启动命令加载失误了，已修复

liyiecho, 2019-12-06:
1.7-18版可以正常控制了，但还是有个小问题，当从停用状态到启用状态时`AdGuardHome.AdGuardHome.enabled='1'`，`AdGuardHome.AdGuardHome.old_enabled`应该为0，可现在却是1
[image]

rufengsuixing, 2019-12-06:
old_enabled 是为了控制重定向功能的，会检测两次开关状态来决定是否改动防火墙或者dnsmasq规则，因为现在我没有实现快速reload，可能用处少一点

rufengsuixing, 2019-12-06:
已经实现了快速reload

Mosney, 2019-12-06:
1.7-21测试正常