rufengsuixing/luci-app-adguardhome #14: 关于AdGuardHome的统计数据在路由重启后被重置

garypang13, 2019-12-04:
在root/etc/init.d/AdGuardHome中找到
		if [ ! -L $workdir/data/stats.db ]; then
		mv -f $workdir/data/stats.db /tmp/stats.db
		ln -s /tmp/stats.db $workdir/data/stats.db
		fi
这样把stats.db目录改到/tmp下,会导致每次重启路由AdGuardHome的统计数据就会被重置,这个数据的保存时间在后台就可以设定,不需要这样强制改为每次重启后重置,建议删除这几行代码.

rufengsuixing, 2019-12-04:
你删了你就启动不了了看readme的已知问题相关

garypang13, 2019-12-04:
> 你删了你就启动不了了看readme的已知问题相关，请自行修改工作目录

我测试过了,删除后能正常启动,

rufengsuixing, 2019-12-04:
输出你的mount命令结果，和你的工作目录

garypang13, 2019-12-04:
[image]
[image]

rufengsuixing, 2019-12-04:
用最新的release尝试一下

garypang13, 2019-12-04:
可以了,感谢!

closed 2019-12-04 by garypang13

Mosney, 2019-12-05:
@garypang13 请问你上面截图用的什么软件