rufengsuixing/luci-app-adguardhome #16: 建议路径改成如:/usr/bin/adg,目前路径太长，想单独编译二进制进去失败！改成短路径成功！

zxlhhyccc, 2019-12-04:
[image]

rufengsuixing, 2019-12-04:
是不是路径所在的地方被文件夹占了位置导致失败

liyiecho, 2019-12-04:
可以参考这个
https://github.com/happyzhang1995/openwrt-adguardhome/blob/master/Makefile

rufengsuixing, 2019-12-04:
这个编译问题我不知道该怎么解决，最好是能给出编译失败后，/usr/bin/AdGuardHome这个文件夹？文件？的状态

rufengsuixing, 2019-12-04:
个人判断，路径长度不是问题，一般是位置被其他类型文件占用

rufengsuixing, 2019-12-04:
本luci没有在/usr/bin产生编译时会产生的文件夹

rufengsuixing, 2019-12-04:
另外我建议你不用在生成固件时打包，如果adguardhome主程序更新，你将需要重新刷入主程序，要不然会占用更多的空间，本luci提供的压缩功能能提供差不多级别的主程序压缩

closed 2019-12-04 by rufengsuixing

zxlhhyccc, 2019-12-04:
> 另外我建议你不用在生成固件时打包，如果adguardhome主程序更新，你将需要重新刷入主程序，要不然会占用更多的空间，本luci提供的压缩功能能提供差不多级别的主程序压缩

我明白了什么情况了，我是同时编译了adguardhome二进制在usr/bin/目录下，而同时又生成与二进制文件同名的文件夹，无法创建造成。。。。所以建议文件夹改为其他名字，谢谢！！

zxlhhyccc, 2019-12-04:
> 另外我建议你不用在生成固件时打包，如果adguardhome主程序更新，你将需要重新刷入主程序，要不然会占用更多的空间，本luci提供的压缩功能能提供差不多级别的主程序压缩

我这样做的的目的是acc可以使用/usr/bin下的二进制，luci又可以使用adguardhome目录下的二进制，主要是我的K3因为阉割了的原因必须手工编译一个二进制文件，如果用luci下载的二进制是不能运行的。。。。

Mosney, 2019-12-05:
> > 另外我建议你不用在生成固件时打包，如果adguardhome主程序更新，你将需要重新刷入主程序，要不然会占用更多的空间，本luci提供的压缩功能能提供差不多级别的主程序压缩
> 
> 我明白了什么情况了，我是同时编译了adguardhome二进制在usr/bin/目录下，而同时又生成与二进制文件同名的文件夹，无法创建造成。。。。所以建议文件夹改为其他名字，谢谢！！

你这个要求就太无脑了