rufengsuixing/luci-app-adguardhome #20: 新功能讨论投票

rufengsuixing, 2019-12-08:
AdH beta 版本支持（emmm不是很难，但是一旦不稳定这个锅emmm）

rufengsuixing, 2019-12-08:
替换dnsmasq(AdG没有继承dnsmasq配置的话已做到,如果继承配置没有ipset也做不到完整)，串联以dnsmasq为上游（能做到，然而老哥你还是手动设置AdG dns上游为dnsmasq，万一脚本设错了就2333）

rufengsuixing, 2019-12-08:
加入文件目标为文件夹或者相反情况的警告检测（警告不应用设置还是做得到，因为应用设置会无法启动，如果自动用删除解决问题可能会有严重后果，万一是/岂不gg）

rufengsuixing, 2019-12-08:
配置切换目录或者路径时移动文件（包含日志，程序等，如果目标不存在的话，然而也会有问题，万一就想建立新文件呢，如果和上一条合并了情况就更复杂了）

rufengsuixing, 2019-12-08:
gfwlist列表的快速删除（实际上没啥重要性，手动也可以删）

rufengsuixing, 2019-12-08:
手动配置编辑器选择时看不清（主题问题，难道要动态更新下编辑器主题，再想想）

rufengsuixing, 2019-12-08:
重定向53端口到AdH中ipv6的部分改为使用tproxy而不是nat redirect(反正都需要额外安装，相比之间tproxy或许更优秀)

Mosney, 2019-12-09:
上面这些功能无感，只希望能保持前向兼容性。
现有功能里`gfw列表上游服务器这个`功能感觉很鸡肋，没想到什么应用场景，爬墙客户端自然会处理掉DNS污染问题。
另外替换掉dnsmasq风险太高，大概率造成未知问题，尤其是小白用户，动手能力强的也不需要luci-app来完成这一操作。

rufengsuixing, 2019-12-09:
luci-app要做到的主要就是自动快速，可以替代繁杂的手动设置，所有的手动都能做到，但是是点一下确认还是各种页面切换半天也是不一样的体验，替换掉dnamsq风险实际不大，不会导致不能连上路由器这种严重后果，最多只是dns无效，在设置里可以轻松调回

jkle112, 2019-12-10:
主要是adguardhome功能不够强大，还需要增加更多的安全功能。

Mosney, 2019-12-10:
> 主要是adguardhome功能不够强大，还需要增加更多的安全功能。

比如说？

closed 2019-12-10 by rufengsuixing