rufengsuixing/luci-app-adguardhome #21: 使用53端口替换dnsmasq失败

garypang13, 2019-12-08:
一直用的重定向53端口到AdGuardHome,使用正常,
改为使用53端口替换dnsmasq后所有网站都打不开解析失败了,不知道这个功能是否还需要额外修改openwrt其他配置

rufengsuixing, 2019-12-08:
需要修改dns监听ip为0.0.0.0

garypang13, 2019-12-08:
AdGuardHome.yaml中DNS监听ip默认就是0.0.0.0
dns:
  bind_host: 0.0.0.0
  port: 5335
依然无效.

rufengsuixing, 2019-12-08:
是因为以前版本残留的ipt规则导致的，这个我在新版中已经做了修复，你可以先重定向切换为无，然后手动重启防火墙或者路由器，再开启

garypang13, 2019-12-08:
[image]
还是不行,

rufengsuixing, 2019-12-08:
那我也不清楚了，本地测试未能复现你所说的问题，你可以将详细配置贴在下面

rufengsuixing, 2019-12-08:
如果使用了ipv6,重定向53端口到AdGuardHome功能大概率是无效的，可能是你的adg配置根本就是错的

rufengsuixing, 2019-12-08:
可以进网页看是否有请求

garypang13, 2019-12-08:
并没有使用Ipv6, adguardhome查询日志中没有请求,
adg配置:
`bind_host: 0.0.0.0
bind_port: 100
users:
- name: root
  password: $2a$10$gKp8P6rscf3xos7ycHgZSOCXFt4S3cAKp24.LmSe0nD53UBKrFQb2
language: zh-cn
rlimit_nofile: 0
dns:
  bind_host: 0.0.0.0
  port: 5335
  statistics_interval: 7
  protection_enabled: true
  filtering_enabled: true
  filters_update_interval: 12
  blocking_mode: nxdomain
  blocked_response_ttl: 86400
  querylog_enabled: true
  querylog_interval: 7
  ratelimit: 0
  ratelimit_whitelist: []
  refuse_any: true
  bootstrap_dns:
  - 218.102.23.228:53
  - 119.29.29.29:53
  all_servers: true
  allowed_clients: []
  disallowed_clients: []
  blocked_hosts: []
  parental_block_host: ""
  safebrowsing_block_host: ""
  blocked_services: []
  cache_size: 4194304
  parental_sensitivity: 13
  parental_enabled: false
  safesearch_enabled: false
  safebrowsing_enabled: false
  safebrowsing_cache_size: 1048576
  safesearch_cache_size: 1048576
  parental_cache_size: 1048576
  cache_time: 30
  rewrites: []
  upstream_dns:
  - tcp://208.67.222.222:5353
  - tls://1.1.1.1
  - https://dns.google/dns-query
  - tls://dns.quad9.net
  - tcp://168.95.1.1
  - tcp://202.141.162.123:5353
  - tls://dns.rubyfish.cn
  - tls://sdns.233py.com
  - tcp://202.86.191.50
  - tcp://101.6.6.6:5353
  - tcp://210.0.255.250
  - tcp://203.80.96.10
  - 127.0.0.1:5555
tls:
  enabled: false
  server_name: ""
  force_https: false
  port_https: 443
  port_dns_over_tls: 853
  certificate_chain: ""
  private_key: ""
  certificate_path: ""
  private_key_path: ""
filters:
- enabled: true
  url: https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt
[... 175 lines cut ...]
  use_global_blocked_services: true
  blocked_services: []
- name: phone_ma
  ip: 10.0.0.178
  mac: ""
  use_global_settings: true
  filtering_enabled: false
  parental_enabled: false
  safebrowsing_enabled: false
  safesearch_enabled: false
  use_global_blocked_services: true
  blocked_services: []
- name: phone_jia
  ip: 10.0.0.107
  mac: ""
  use_global_settings: true
  filtering_enabled: false
  parental_enabled: false
  safebrowsing_enabled: false
  safesearch_enabled: false
  use_global_blocked_services: true
  blocked_services: []
log_file: ""
verbose: false
schema_version: 5`

rufengsuixing, 2019-12-08:
使用替换53端口，开启运行日志甚至勾选详细日志，查找问题

garypang13, 2019-12-09:
```

2019/12/09 02:46:13 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to [2001:4860:4860::8844]:443, in 5 milliseconds, cause: dial tcp [2001:4860:4860::8844]:443: socket: address family not supported by protocol
2019/12/09 02:46:13 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to [2001:4860:4860::8844]:443
2019/12/09 02:46:13 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to [2001:4860:4860::8888]:443, in 6 milliseconds, cause: dial tcp [2001:4860:4860::8888]:443: socket: address family not supported by protocol
2019/12/09 02:46:13 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to [2001:4860:4860::8888]:443
2019/12/09 02:46:13 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to 8.8.8.8:443, in 10000 milliseconds, cause: dial tcp 8.8.8.8:443: i/o timeout
2019/12/09 02:46:03 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to [2001:4860:4860::8844]:443, in 5 milliseconds, cause: dial tcp [2001:4860:4860::8844]:443: socket: address family not supported by protocol
2019/12/09 02:46:03 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to [2001:4860:4860::8844]:443
2019/12/09 02:46:03 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to [2001:4860:4860::8888]:443, in 7 milliseconds, cause: dial tcp [2001:4860:4860::8888]:443: socket: address family not supported by protocol
2019/12/09 02:46:03 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to [2001:4860:4860::8888]:443
2019/12/09 02:46:03 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to 8.8.8.8:443, in 10000 milliseconds, cause: dial tcp 8.8.8.8:443: i/o timeout
2019/12/09 02:46:03 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to 8.8.8.8:443
2019/12/09 02:46:03 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to 8.8.4.4:443, in 10000 milliseconds, cause: dial tcp 8.8.4.4:443: i/o timeout
2019/12/09 02:45:53 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to 8.8.8.8:443
2019/12/09 02:45:53 30957#160 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to 8.8.4.4:443
2019/12/09 02:45:53 30957#138 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer failed to initialize connection to 8.8.4.4:443, in 10000 milliseconds, cause: dial tcp 8.8.4.4:443: i/o timeout
2019/12/09 02:45:53 30957#12 [debug] github.com/AdguardTeam/dnsproxy/upstream.lookup(): successfully finished lookup for dns.quad9.net in 10001 milliseconds using 218.102.23.228:53. Result : [{9.9.9.9 } {149.112.112.112 }]
2019/12/09 02:45:51 30957#58 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 149.112.112.112:853 in 7775 milliseconds
2019/12/09 02:45:45 30957#145 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 149.112.112.112:853 in 1913 milliseconds
2019/12/09 02:45:44 30957#104 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 149.112.112.112:853 in 810 milliseconds
2019/12/09 02:45:44 30957#47 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 1.1.1.1:853 in 254 milliseconds
2019/12/09 02:45:43 30957#107 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 118.89.110.78:853 in 168 milliseconds
2019/12/09 02:45:43 30957#148 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 118.89.110.78:853 in 163 milliseconds
2019/12/09 02:45:43 30957#149 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 119.29.107.85:853 in 159 milliseconds
2019/12/09 02:45:43 30957#108 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): dialer has successfully initialized connection to 119.29.107.85:853 in 161 milliseconds
2019/12/09 02:45:43 30957#25 [debug] No answer for rDNS lookup of 10.0.0.1

10.IN-ADDR.ARPA.	86400	IN	SOA	10.IN-ADDR.ARPA. . 0 28800 7200 604800 86400
;; AUTHORITY SECTION:

;1.0.0.10.in-addr.arpa.	IN	 PTR
;; QUESTION SECTION:

;; flags: qr aa rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 0
2019/12/09 02:45:43 30957#45 [debug] github.com/AdguardTeam/dnsproxy/proxy.(*Proxy).logDNSMessage(): OUT: ;; opcode: QUERY, status: NXDOMAIN, id: 41349
2019/12/09 02:45:43 30957#45 [debug] github.com/AdguardTeam/dnsproxy/proxy.(*Proxy).Resolve(): RTT: 150 ms
2019/12/09 02:45:43 30957#145 [debug] github.com/AdguardTeam/dnsproxy/upstream.createDialContext.func1(): Dialing to 149.112.112.112:853
[... cut ...]
2019/12/09 02:45:41 29985#49 [debug] github.com/AdguardTeam/AdGuardHome/stats.(*statsCtx).beginTxn(): db.Begin...
2019/12/09 02:45:41 29985#49 [info] Stopped the DNS proxy server
2019/12/09 02:45:41 29985#9 [info] udpListen.ReadFrom() returned because we're reading from a closed connection, exiting loop
2019/12/09 02:45:41 29985#10 [info] tcpListen.Accept() returned because we're reading from a closed connection, exiting loop
2019/12/09 02:45:41 29985#49 [info] Stopping the DNS proxy server
2019/12/09 02:45:41 29985#49 [info] Stopping AdGuard Home 
```

rufengsuixing, 2019-12-09:
没看出什么所以然，不知道你那边什么原因，如果成功替换了53端口，iptables里没有多余的规则，剩余的问题就去adg反馈吧
学会安装bind-dig，然后使用dig对dns服务器进行debug

closed 2019-12-09 by rufengsuixing

rufengsuixing, 2019-12-18:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/d45e97b709de826779cfe7e5f99c4454ecf4ccd8