rufengsuixing/luci-app-adguardhome #22: 这样正常吗？？

adaaaaaa, 2019-12-08:
上次检测到的主程序版本: no config no bin

rufengsuixing, 2019-12-08:
已经尝试下载了，然而dns有问题失败了，日志中不是有
Resolving github.com... failed: Try again.

rufengsuixing, 2019-12-08:
另一个是程序没启动没日志的时候尝试打开日志会报错

rufengsuixing, 2019-12-08:
我会试着用其他的dns进行解析

rufengsuixing, 2019-12-08:
执行文件下载需要勾选enable

rufengsuixing, 2019-12-08:
请告诉我，你勾选的重定向设置

adaaaaaa, 2019-12-08:
> 请告诉我，你勾选的重定向设置

没有做任何设置啊。。。都是默认的。。。
DNS好像没问题也。。。诊断能ping通。。。

rufengsuixing, 2019-12-08:
github.com确实有被干扰的现象
另外你给的信息太少，我无法判断问题，
如果出现这个的时候是开机时，可能是启动顺序早于了dnsmasq，然而现在并不能确定

rufengsuixing, 2019-12-08:
wget 不能用其他dns进行解析，所以只能加多尝试次数

adaaaaaa, 2019-12-08:
> 可能是启动顺序早于了dnsmasq

开机顺序也会导致问题？？
你还需要知道什么信息？

rufengsuixing, 2019-12-08:
如果不能复现就算了吧，有的dns第一次查询确实会超时，之后就正常了

adaaaaaa, 2019-12-08:
Sun Dec  8 20:24:22 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Dec  8 20:24:22 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

Sun Dec  8 20:24:24 2019 daemon.notice procd: /etc/rc.d/S95AdGuardHome: awk: /etc/AdGuardHome.yaml: No such file or directory
Sun Dec  8 20:24:24 2019 daemon.notice procd: /etc/rc.d/S95AdGuardHome: uci: Entry not found

这样的启动顺序是你说的那样吧？？

> 可能是启动顺序早于了dnsmasq

rufengsuixing, 2019-12-08:
这不明显在dnsmasq后启动的

rufengsuixing, 2019-12-08:
为什么不试着下载核心呢

adaaaaaa, 2019-12-08:
> 为什么不试着下载核心呢

是这个吗？？
git clone https://github.com/AdguardTeam/AdGuardHome.git
搞了两个固件，一个有，一个没有这个。。。

rufengsuixing, 2019-12-08:
那么大个按钮

rufengsuixing, 2019-12-08:
你开心就好

adaaaaaa, 2019-12-08:
> 那么大个按钮

更新核心吗？？

 Local version: ., cloud version: v0.99.3.
Updating core...
start download v0.99.3/AdGuardHome_linux_amd64.tar.gz
Failed to download core.

rufengsuixing, 2019-12-08:
下载失败没什么好说的，找原因去吧

rufengsuixing, 2019-12-08:
脚本位置在/usr/bin/AdGuardHome/update_core.sh

adaaaaaa, 2019-12-08:
好像成功了。。。谢谢你，打扰你了。。。🤪🤪🤪

adaaaaaa, 2019-12-08:
怎么手动添加规则给路由器里面的adguard呢？
it之家的广告去的不够彻底。。。
ublock就很彻底。。。

adaaaaaa, 2019-12-08:
规则和ublock的通用吗？加上是不是就可以解决了？？😂

rufengsuixing, 2019-12-08:
https://github.com/rufengsuixing/luci-app-adguardhome/issues/6

closed 2019-12-08 by adaaaaaa