rufengsuixing/luci-app-adguardhome #23: 重启后规则自动下载会在PPPoE完成前执行

Mosney, 2019-12-09:
因为用的U盘做系统盘且工作目录包含查询记录、数据库等频繁写入的文件，为了避免U盘老化，故把工作目录设定到了/var，也就是tmpfs内存里。这样重启会丢失所有数据，理论上ADG发现规则文件不存在即会自动下载，但是路由重启之后需要一点时间接通网络（特别是家用PPPoE），ADG本身只尝试一次，网络错误之后就不再次尝试，因此导致出现过滤真空期，只有到下次自动更新时才能正常运作。 

作为luci-app似乎不能控制这一动作，只能加一个延时或者侦测到日志输出规则文件不存在且网络错误时自动重启。

rufengsuixing, 2019-12-09:
这个确实超了luci的范围，但是你可以自行写一个init脚本，设置启动优先级高于adg，在开机时将你备份的规则文件复制到var下

rufengsuixing, 2019-12-09:
我也会考虑加入延迟启动个能

Mosney, 2019-12-09:
> 我也会考虑加入延迟启动功能

我也在ADG那边开了个issue，如果他们能加重试的话这边就不用改了。

rufengsuixing, 2019-12-09:
我有两种解决方案，一种是仅对开机有效，会检测到网络连接再继续。第二种是绑定网卡up的动作，发现网卡up就重启ad

Mosney, 2019-12-09:
前一种好一点，后一种可能会因为ISP的超时强制断线等原因导致无谓的重启。

rufengsuixing, 2019-12-09:
如果像我现在这样改，你如果想使用AdH作为53端口，必须需要使用替换dnsmasq这个选项，否则在开机时ping会因为dns的原因不会成功的，会延迟3分钟启动

rufengsuixing, 2019-12-09:
我试着加入了百度的ip和8.8.8.8的ping，希望能够解决没有dns可用的时候无法启动

rufengsuixing, 2019-12-09:
你可以测试一下pre release

Mosney, 2019-12-09:
测试IP设计成这种文本框更好一点？
[image]

Mosney, 2019-12-09:
测试了一下，开机之后一直不启动，运行的进程如下图
[image]

rufengsuixing, 2019-12-10:
我觉得脚本没问题，虽然luci界面默认是勾选的但是需要保存一次才会生效

rufengsuixing, 2019-12-10:
你可以检查/root/etc/init.d/AdGuardHome 中 boot里的相关脚本

rufengsuixing, 2019-12-10:
fix

Mosney, 2019-12-10:
确认没问题了

closed 2019-12-10 by Mosney