rufengsuixing/luci-app-adguardhome #25: 1.7-28release把luci搞坏了

YujiaCheng1996, 2019-12-13:
### 问题详细信息

* **AdGuard Home luci的版本:**
  * 1.7-28
* **AdGuard Home server的版本:**
  * 0.99.3
* **操作系统和版本:**
  * PandoraBox armv7

### 实际的执行结果
不仅仅是对应页，而是整个web后台都无法打开，页面提示
/usr/lib/lua/luci/dispatcher.lua:431: Invalid controller file found
The file '/usr/lib/lua/luci/controller/adguardhome.lua' contains an invalid module line.
Please verify whether the module name is set to 'luci.controller.adguardhome' - It must correspond to the file path!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:431: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:188: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

### 更多的信息
安装时提示：
#opkg install luci-app-adguardhome_1.7-28_all.ipk 
Upgrading luci-app-adguardhome on root from 1.7-27 to 1.7-28...
AdGuardHome turn off
enabled=0
Configuring luci-app-adguardhome.
uci: Entry not found
uci: Entry not found
uci: Entry not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/AdGuardHome is different from the conffile in the new package. The new conffile will be placed at /etc/config/AdGuardHome-opkg.
这个问题回退安装1.7-27版本即恢复正常。

rufengsuixing, 2019-12-13:
不好意思打包的时候文件重复了，文件已更新重新下release即可

YujiaCheng1996, 2019-12-14:
新release已解决，谢谢

closed 2019-12-14 by YujiaCheng1996