rufengsuixing/luci-app-adguardhome #28: 建议增加自动更新主程序选项

garypang13, 2019-12-15:
建议增加自动更新主程序选项,
可参考
https://github.com/maxlicheng/luci-app-unblockmusic/blob/1bb55ac62762f3f2603f1f84d2d32501c18244d3/app/root/etc/init.d/unblockmusic
```
add_cron()
{
	if [ $AUTOUPDATE -eq 1 ]; then
    sed -i '/update_core.sh/d' $CRON_FILE
		echo '0 2 * * * /usr/share/AdGuardHome/update_core.sh 2>&1' >> $CRON_FILE
		crontab $CRON_FILE
	fi
}

del_cron()
{
	sed -i '/update_core.sh/d' $CRON_FILE
	/etc/init.d/cron restart
}
```

rufengsuixing, 2019-12-15:
计划任务也不能保证你一定能及时更新，另外adh没到1.0版本，不敢说接口会不会有变动，自动升级炸了也难说

rufengsuixing, 2019-12-15:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/df9b68fb05c6ee425548646c1429f35ac5a1ed85

Mosney, 2019-12-16:
这个功能完全没必要，ADG本就支持自我升级，只需要在WebUI里点一点就可以

rufengsuixing, 2019-12-16:
功能加了，默认凌晨3：30，对时间不满意可以自行调节，只要不手动关闭自动更新，不会更改/删除crontab的时间记录

closed 2019-12-16 by rufengsuixing