rufengsuixing/luci-app-adguardhome #29: 显示错位，无法启动

happyzhang1995, 2019-12-15:
[image]

OpenWrt R9.12.03 / LuCI Master (git-19.338.43082-9e87e69)
--
4.19.88

检查更新一直失败
 
Local version: ., cloud version: v0.99.3.
Updating core...
start download v0.99.3/AdGuardHome_linux_amd64.tar.gz
--2019-12-15 21:11:18--  https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.99.3/AdGuardHome_linux_amd64.tar.gz
Resolving github.com... 13.250.177.223
Connecting to github.com|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/e89f6880-0ade-11ea-9de1-7c5d37f46686?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191215%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191215T131120Z&X-Amz-Expires=300&X-Amz-Signature=674548e55d35b5533552565b21bc2186311b216ca30be065a4ab48345aa6b2cc&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_amd64.tar.gz&response-content-type=application%2Foctet-stream [following]
--2019-12-15 21:11:22--  https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/e89f6880-0ade-11ea-9de1-7c5d37f46686?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191215%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191215T131120Z&X-Amz-Expires=300&X-Amz-Signature=674548e55d35b5533552565b21bc2186311b216ca30be065a4ab48345aa6b2cc&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_amd64.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com... 52.216.16.40
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|52.216.16.40|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6383323 (6.1M) [application/octet-stream]
Saving to: '/tmp/AdGuardHomeupdate/AdGuardHome_linux_amd64.tar.gz'

     0K                                                         0% 0.00 =0s

2019-12-15 21:11:45 (0.00 B/s) - Read error at byte 0/6383323 (No error information). Retrying.

--2019-12-15 21:11:46--  (try: 2)  https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/e89f6880-0ade-11ea-9de1-7c5d37f46686?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191215%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191215T131120Z&X-Amz-Expires=300&X-Amz-Signature=674548e55d35b5533552565b21bc2186311b216ca30be065a4ab48345aa6b2cc&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_amd64.tar.gz&response-content-type=application%2Foctet-stream
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|52.216.16.40|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6383323 (6.1M) [application/octet-stream]
Saving to: '/tmp/AdGuardHomeupdate/AdGuardHome_linux_amd64.tar.gz'

     0K .......... ......                                       0% 3.17K=5.2s

2019-12-15 21:12:15 (3.17 KB/s) - Read error at byte 16959/6383323 (No error information). Giving up.

download failed
tar: unexpected end of file
tar: short read
download success start copy
start copy
awk: /etc/AdGuardHome.yaml: No such file or directory
AdGuardHome turn off
enabled=0
Command failed: Not found
rm: can't remove '/usr/bin/AdGuardHome/AdGuardHome': No such file or directory
mv: can't rename '/tmp/AdGuardHomeupdate/AdGuardHome/AdGuardHome': No such file or directory
mv failed maybe not enough space please use upx or change bin to /tmp/AdGuardHome
uci: Entry not found

A task is already running.

rufengsuixing, 2019-12-15:
下载失败，检查你和github的连接

rufengsuixing, 2019-12-15:
另外你把我脚本中exit 都删了也是个猛人

happyzhang1995, 2019-12-15:
> 另外你把我脚本中exit 都删了也是个猛人

我直接git clone，完全没有改你的脚本

rufengsuixing, 2019-12-15:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/005dffca2e48f632ba3fe38deec71860488223a3

happyzhang1995, 2019-12-16:
> [005dffc](https://github.com/rufengsuixing/luci-app-adguardhome/commit/005dffca2e48f632ba3fe38deec71860488223a3)

可以使用了，但我不太明白为什么我有些字体是问号

closed 2019-12-16 by happyzhang1995

rufengsuixing, 2019-12-16:
这个问号代表你没有配置文件，有了文件端口获取就正常了