rufengsuixing/luci-app-adguardhome #33: 关机时备份工作目录 开机后未恢复

Mosney, 2019-12-16:
关机前备份工作目录正常，但是备份文件在开机后没有正确恢复到工作目录。
看了一下应该是恢复的时候少了一层`/data`目录
```
root@OP19:~# tree /usr/lib/AdGuardHomeData/
/usr/lib/AdGuardHomeData/
└── data
    ├── filters
    │   ├── 1.txt
    │   ├── 1575632085.txt
    │   ├── 2.txt
    │   ├── 3.txt
    │   └── 4.txt
    ├── querylog.json
    ├── sessions.db
    └── stats.db

2 directories, 8 files
root@OP19:~# tree /tmp/AdGuardHome
/tmp/AdGuardHome
├── data
│   ├── filters
│   │   ├── 1.txt
│   │   ├── 1575632085.txt
│   │   ├── 2.txt
│   │   ├── 3.txt
│   │   └── 4.txt
│   ├── sessions.db
│   └── stats.db
├── filters
│   ├── 1.txt
│   ├── 1575632085.txt
│   ├── 2.txt
│   ├── 3.txt
│   └── 4.txt
├── querylog.json
├── sessions.db
└── stats.db

3 directories, 15 files

```

Mosney, 2019-12-16:
https://github.com/rufengsuixing/luci-app-adguardhome/blob/3091c89c70eb3f8d0c28334fc36e3f763708db4d/root/etc/init.d/AdGuardHome#L346
https://github.com/rufengsuixing/luci-app-adguardhome/blob/3091c89c70eb3f8d0c28334fc36e3f763708db4d/root/etc/init.d/AdGuardHome#L450

Mosney, 2019-12-16:
直接运行上面两条命令都按预期执行 但是开机启动的确是会复制到错误的位置

rufengsuixing, 2019-12-17:
应该没毛病了

closed 2019-12-17 by rufengsuixing

Mosney, 2019-12-17:
问题依旧

rufengsuixing, 2019-12-17:
请重新在界面点击一次提交，会自动去掉路径尾部的/

Mosney, 2019-12-18:
没有任何luci配置的路径最后带/，也重新提交了一遍配置，重启复制依然少一层/data

rufengsuixing, 2019-12-18:
本地测试没有复现，是不是以前的错误拷贝没有删除

Mosney, 2019-12-20:
问题依旧，开机后复制的目录依然少一层`/data`，另外正常运行一段时间`querylog.json`这个文件会非常大，并且完整的查询记录大部分情况下没太有什么意义，可以考虑默认不备份。
[image]

rufengsuixing, 2019-12-20:
有日志截断功能

rufengsuixing, 2019-12-20:
cp很有趣，在目标文件夹不存在的时候cp的执行结果不同
如果存在则粘贴文件夹到路径下
不存在则粘贴文件夹为目标文件夹

Mosney, 2019-12-20:
> 直接运行上面两条命令都按预期执行 但是开机启动的确是会复制到错误的位置

哈哈哈 看来我应该进一步测试一下确认是cp的锅的

rufengsuixing, 2019-12-20:
周末暂时没时间等周一我会考虑改成多选框

Mosney, 2019-12-20:
> 有日志截断功能

目前截断阈值是这个文件增长到多大?

rufengsuixing, 2019-12-20:
crontab自行调整，默认每小时截到2000行