rufengsuixing/luci-app-adguardhome #34: Unable to get AdGuardHome luci working on arm

zeus10000, 2019-12-17:
### Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of luci-app-adguardhome
- [ ] I checked to make sure that this issue has not already been filed

### Issue Details

* **Version of AdGuard Home luci:**
  * 
* **Version of AdGuard Home server:**
  * 
* **Operating system and version:**
  * 

### Expected Behavior

### Actual Behavior

### Screenshots

<details><summary>Screenshot:</summary>

</details>

### Additional Information

rufengsuixing, 2019-12-17:
the version of your wget don't support https ,you may use busybox's wget ,try opkg install wget，then you will get wget-ssl to use,maybe next version i will use curl instead

zeus10000, 2019-12-17:
Yes, I am using busybox's wget. I had to remove the default wget and reinstall and it worked.
```
root@xxxxx:~# opkg install wget
Installing wget (1.19.5-3) to root...
Downloading http://downloads.openwrt.org/releases/18.06.5/packages/arm_cortex-a9_vfpv3/packages/wget_1.19.5-3_arm_cortex-a9_vfpv3.ipk
Collected errors:
 * check_data_file_clashes: Package wget wants to install file /usr/bin/wget
        But that file is already provided by package  * wget-nossl
 * opkg_install_cmd: Cannot install package wget.
root@xxxxx:~# opkg remove  wget-nossl
No packages removed.
Collected errors:
 * print_dependents_warning: Package wget-nossl is depended upon by packages:
 * print_dependents_warning:    luci-app-adguardhome
 * print_dependents_warning: These might cease to work if package wget-nossl is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
root@xxxxx:~# opkg remove  wget-nossl  --force-depends
Removing package wget-nossl from root...
root@xxxxx:~# opkg install wget
Installing wget (1.19.5-3) to root...
Downloading http://downloads.openwrt.org/releases/18.06.5/packages/arm_cortex-a9_vfpv3/packages/wget_1.19.5-3_arm_cortex-a9_vfpv3.ipk
Configuring wget.
root@xxxxx:~# wget https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.99.3/AdGuardHome_linux_arm.tar.gz
--2019-12-16 10:25:27--  https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.99.3/AdGuardHome_linux_arm.tar.gz
Resolving github.com... 140.82.114.4
Connecting to github.com|140.82.114.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/edfcb300-0ade-11ea-9a17-24e7bd628613?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191217%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191217T093838Z&X-Amz-Expires=300&X-Amz-Signature=a52f3fb476df736e414e0aa87ded6f98bbd82252f63b7907aa6361fa675badb1&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_arm.tar.gz&response-content-type=application%2Foctet-stream [following]
--2019-12-16 10:25:27--  https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/edfcb300-0ade-11ea-9a17-24e7bd628613?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191217%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191217T093838Z&X-Amz-Expires=300&X-Amz-Signature=a52f3fb476df736e414e0aa87ded6f98bbd82252f63b7907aa6361fa675badb1&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_arm.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com... 52.216.144.196
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|52.216.144.196|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6048641 (5.8M) [application/octet-stream]
Saving to: 'AdGuardHome_linux_arm.tar.gz'

AdGuardHome_linux_arm.tar.gz                         100%[===================================================================================================================>]   5.77M  11.6MB/s    in 0.5s

2019-12-16 10:25:28 (11.6 MB/s) - 'AdGuardHome_linux_arm.tar.gz' saved [6048641/6048641]
```

Changed permission to 777 for /usr/bin/AdGuardHome directory and AdGuardHome bin file and it recognized the file

```
root@xxxxx:/usr/bin# ll AdGuardHome/
drwxrwxrwx    3 root     root           296 Dec 16 10:37 ./
drwxr-xr-x    1 root     root           432 Dec 16 10:33 ../
-rwxrwxrwx    1 root     root      13126088 Dec 16 10:35 AdGuardHome*
drwxr-xr-x    3 root     root           368 Dec 16 10:37 data/
```

[image]

closed 2019-12-17 by rufengsuixing