rufengsuixing/luci-app-adguardhome #36: 前面两个修改的版本不能运行，现在更新一下到1.7.32版本试试能不能运行！

zxlhhyccc, 2019-12-17:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [ ] 我用着最新版的 luci-app-adguardhome
- [ ] 我检查了没有人提交同样的问题

### 问题描述

### 新功能描述

### 你能想到的解决方法

### 更多的信息

zxlhhyccc, 2019-12-17:
[image]

zxlhhyccc, 2019-12-17:
更新到1.7-32版本也是不能运行，降级到1.7.13版本去。。。。

rufengsuixing, 2019-12-17:
你自行使用了/usr/bin/AdGuardHome这个文件路径，就要注意，后面有很多文件夹路径默认是这个，需要手动改掉，不然无法提交

closed 2019-12-17 by rufengsuixing

zxlhhyccc, 2019-12-17:
> 你自行使用了/usr/bin/AdGuardHome这个文件路径，就要注意，后面有很多文件夹路径默认是这个，需要手动改掉，不然无法提交

这个我知道，后面的我都做了相应的路径修改，我现在降级到1.7.13版本正常运行。。。。

rufengsuixing, 2019-12-17:
至少提供adg启动的日志，不然无法判断问题

rufengsuixing, 2019-12-17:
或着手动启动/etc/init.d/AdGuardHome 时的信息

zxlhhyccc, 2019-12-17:
日志没显示，显示的是na/na。、/na。。。。

zxlhhyccc, 2019-12-17:
现在我编译了1.7.13版本，卸载后无法安装新版本，不知道什么原因。。。

rufengsuixing, 2019-12-17:
你没有设置日志显示，怎么可能显示日志，给日志指定路径或者syslog

zxlhhyccc, 2019-12-18:
> 你没有设置日志显示，怎么可能显示日志，给日志指定路径或者syslog

晚上来测试一下你说的1.7.33版本

rufengsuixing, 2019-12-18:
请不要滥用issue