rufengsuixing/luci-app-adguardhome #37: 我已经编译了最新版本的adg（0.100.1），请问如何将luci的下载版本改成最新的？我可以放在仓库上！

zxlhhyccc, 2019-12-17:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [ ] 我用着最新版的 luci-app-adguardhome
- [ ] 我检查了没有人提交同样的问题

### 问题描述

### 新功能描述

### 你能想到的解决方法

### 更多的信息

rufengsuixing, 2019-12-17:
暂无计划支持自定义版本，可以使用winscp自行上传到路由器，文件被文件夹位置所占，或者相反会导致很多无法预测的异常

rufengsuixing, 2019-12-18:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/008217711202524ad9531a0a4531e2dd920e5ee6

rufengsuixing, 2019-12-18:
尝试prerelease

zxlhhyccc, 2019-12-18:
> 尝试prerelease
prerelease是什么版本？

zxlhhyccc, 2019-12-18:
prerelease这个版本源码还没放出来吧？

> 尝试prerelease

zxlhhyccc, 2019-12-18:
晚上来测试一下你说的1.7.33版本

zxlhhyccc, 2019-12-18:
> 晚上来测试一下你说的1.7.33版本
1.7.33版本运行正常，期待放出源码，谢谢！

rufengsuixing, 2019-12-18:
项目是最新源码早就放出来了，是比release早更新的

zxlhhyccc, 2019-12-18:
> 项目是最新源码早就放出来了，是比release早更新的

现在的项目是1.7.32版本，我测试的是1.7.33版本，是不是没更新？

zxlhhyccc, 2019-12-18:
> 项目是最新源码早就放出来了，是比release早更新的

每次应用保存，出现这个错误提示（1.7.33版本也是一样），需要点击右上角的配置保存并应用：
[image]

[image]
[image]

zxlhhyccc, 2019-12-18:
> 尝试prerelease

经测试，1.7.32版本只要编译进去就不会运行。。。

zxlhhyccc, 2019-12-18:
日志：
[image]

rufengsuixing, 2019-12-18:
> 加入gfw列表才能正常运行

不可能必须加入才能运行

zxlhhyccc, 2019-12-18:
> > 加入gfw列表才能正常运行
> 
> 不可能，这个功能是独立的

是真的，没添加gfw列表，同时一个或几个空值，只有添加就正常运行了

zxlhhyccc, 2019-12-18:
> > 加入gfw列表才能正常运行
> 
> 不可能，这个功能是独立的

看见你源码更新了，是解决了那个应用并保存的问题吗？

zxlhhyccc, 2019-12-18:
大佬，添加了gfw列表，也是出现：
[image]
请问缺了什么没勾选？，我目前的选择是：
[image]
[image]
[image]

zxlhhyccc, 2019-12-18:
更新后的源码解决了应用并保存问题，但仍未解决不添加gfw列表不能运行的问题。。。。

rufengsuixing, 2019-12-18:
贴上你的config，删掉gfw列表的

zxlhhyccc, 2019-12-18:
#提交就可以直接用的配置模板文件
#a template config can be use with a apply
bind_host: 0.0.0.0
bind_port: 3000
users:
- name: root
  password: $2y$10$FfeQavihMUiXCuJhHuQwy.6EOXDvkXb/S50qI5fXizqarNT/ShhQm
language: zh-cn
rlimit_nofile: 0
web_session_ttl: 720
dns:
  bind_host: 0.0.0.0
  port: 5553
  statistics_interval: 1
  querylog_enabled: true
  querylog_interval: 1
  querylog_memsize: 1000
  protection_enabled: true
  blocking_mode: nxdomain
  blocked_response_ttl: 10
  ratelimit: 0
  ratelimit_whitelist: []
  refuse_any: false
  bootstrap_dns:
  - 1.0.0.1:53
  - 2606:4700:4700::1001:53
  all_servers: true
  allowed_clients:
  - 127.0.0.1
  disallowed_clients: []
  blocked_hosts: []
  parental_block_host: ""
  safebrowsing_block_host: ""
  cache_size: 4194304
  upstream_dns:
  - 202.141.178.13:5353
  - 202.38.93.153:5353
  - 202.141.162.123:5353
  - https://neatdns.ustclug.org/resolve
  - https://dns.rubyfish.cn/dns-query
  - https://ndns.233py.com/dns-query
  - https://edns.233py.com/dns-query
  - https://wdns.233py.com/dns-query
  - https://sdns.233py.com/dns-query
  - 1.1.1.1
  - 9.9.9.9
  - 117.50.10.10
  - 119.29.29.29
  - 114.114.114.114
  - 223.5.5.5
  - 208.67.222.222:5353
  - 2606:4700:4700::1111
  - 2606:4700:4700::1001
  - tcp://8.8.8.8
  - tcp://8.8.4.4
  - tcp://2001:4860:4860::8888
  - tcp://2001:4860:4860::8844
  - tls://dns.google
  - tls://1dot1dot1dot1.cloudflare-dns.com
  - https://dns9.quad9.net/dns-query
  filtering_enabled: true
  filters_update_interval: 24
  parental_sensitivity: 13
  parental_enabled: false
  safesearch_enabled: false
  safebrowsing_enabled: false
  safebrowsing_cache_size: 1048576
  safesearch_cache_size: 1048576
  parental_cache_size: 1048576
  cache_time: 30
  rewrites: []
  blocked_services: []
tls:
  enabled: false
  server_name: ""
  force_https: false
  port_https: 443
  port_dns_over_tls: 853
  certificate_chain: ""
  private_key: ""
  certificate_path: ""
  private_key_path: ""
filters:
- enabled: true
  url: https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt
  name: AdGuard Simplified Domain Names filter
  id: 1
- enabled: true
  url: https://adaway.org/hosts.txt
  name: AdAway
  id: 2
- enabled: true
  url: https://hosts-file.net/ad_servers.txt
  name: hpHosts - Ad and Tracking servers only
  id: 3
- enabled: true
  url: http://www.malwaredomainlist.com/hostslist/hosts.txt
  name: MalwareDomainList.com Hosts List
  id: 4
- enabled: true
  url: https://raw.githubusercontent.com/523860169/list/master/ad.txt
  name: Diablo AdFilters
  id: 1553073381
- enabled: true
  url: http://tools.yiclear.com/ChinaList2.0.txt
  name: ChinaListV2.0[20190224001]
  id: 1553073382
- enabled: true
  url: https://easylist-downloads.adblockplus.org/antiadblockfilters.txt
  name: Adblock Warning Removal List
  id: 1553073383
- enabled: true
  url: https://fanboy.co.nz/fanboy-problematic-sites.txt
  name: Fanboy's problematic-sites
  id: 1553073384
- enabled: true
  url: https://raw.githubusercontent.com/xinggsf/Adblock-Plus-Rule/master/ABP-FX.txt
  name: "5"
  id: 1553073385
- enabled: true
  url: https://raw.githubusercontent.com/adbyby/xwhyc-rules/master/lazy.txt
  name: "6"
  id: 1553073386
- enabled: true
  url: https://hosts.nfz.moe/127.0.0.1/full/hosts
  name: "7"
  id: 1553073387
- enabled: true
  url: https://raw.githubusercontent.com/vokins/yhosts/master/hosts
  name: "8"
  id: 1553073388
- enabled: true
  url: http://sub.adtchrome.com/adt-chinalist-easylist.txt
  name: ChinaList+EasyList(淇)
  id: 1553073405
- enabled: true
  url: https://easylist.to/easylist/easylist.txt
  name: EasyList
  id: 1568716204
- enabled: true
  url: https://easylist-downloads.adblockplus.org/easylistchina.txt
  name: EasyList China
  id: 1568716205
- enabled: true
  url: https://raw.githubusercontent.com/cjx82630/cjxlist/master/cjx-annoyance.txt
  name: CJX's Annoyance List
  id: 1568716206
- enabled: true
  url: https://raw.githubusercontent.com/cjx82630/cjxlist/master/cjxlist.txt
  name: CJX's EasyList Lite
  id: 1568716207
user_rules: []
dhcp:
  enabled: false
  interface_name: ""
  gateway_ip: ""
  subnet_mask: ""
  range_start: ""
  range_end: ""
  lease_duration: 86400
  icmp_timeout_msec: 1000
clients: []
log_file: ""
verbose: false
schema_version: 6

rufengsuixing, 2019-12-18:
配置貌似没问题，你可以打开运行日志看看，或着按照bug issue的模板填写log日志

zxlhhyccc, 2019-12-18:
> 配置貌似没问题，你可以打开运行日志看看，或着按照bug issue的模板填写log日志

**贴上运行日志：**

[AdGuardHome12-19-1_17.log](https://github.com/rufengsuixing/luci-app-adguardhome/files/3979667/AdGuardHome12-19-1_17.log)

zxlhhyccc, 2019-12-18:
> 配置貌似没问题，你可以打开运行日志看看，或着按照bug issue的模板填写log日志

运行日志貌似没有问题，同时，停用luci后gfw列表自动消失。。。。

rufengsuixing, 2019-12-18:
在下面自动隐藏了，因为有人说功能不常用

rufengsuixing, 2019-12-18:
如果你说的是配置文件无法修改，请使用df关注一下你的存储空间

zxlhhyccc, 2019-12-18:
> 如果你说的是配置文件无法修改，请使用df关注一下你的存储空间

配置文件可以修改，但配置文件你也认为貌似没有问题。。。。

zxlhhyccc, 2019-12-18:
这是我的存储空间，存储空间还有。。。。
[image]

rufengsuixing, 2019-12-18:
贴log当然要贴出问题的的时候的log

zxlhhyccc, 2019-12-18:
> 贴log当然要贴出问题的的时候的log，你说不加不能启动，那就贴不加的时候的log
不加不启动运行日志一片空白，勾选了详细日志。。。

[image]

rufengsuixing, 2019-12-18:
那就只能init_trace了

zxlhhyccc, 2019-12-18:
> init_trace

init_trace是什么意思？只能添加gfw列表后运行吗？

rufengsuixing, 2019-12-19:
请重新开issue并按照模板填写，另外请起有意义有标识性的标题，这里不是qq和微信，不是聊天的地方，以后发现无视模板直接提难以发现的bug的我会直接关闭甚至删除

closed 2019-12-19 by rufengsuixing