rufengsuixing/luci-app-adguardhome #42: 无法启动

PoiYuudachi, 2019-12-19:
OpenWrt R9.12.03  保存并启用还是未运行  ssh里用命令可以运行成功

zxlhhyccc, 2019-12-19:
> OpenWrt R9.12.03 保存并启用还是未运行 ssh里用命令可以运行成功

添加一下gfw列表就可以启动。。。。这是一个bug

rufengsuixing, 2019-12-19:
确认一下，只是界面的提交并应用失效了是么
请列出uci show ucitrack的内容

zxlhhyccc, 2019-12-19:
uci show ucitrack的内容怎么查？

rufengsuixing, 2019-12-19:
命令行输入即可

PoiYuudachi, 2019-12-19:
是的  uci show ucitrack不知道在哪里看  日志是空的

zxlhhyccc, 2019-12-19:
> 命令行输入即可

root@OpenWrt:~# uci show ucitrack
ucitrack.@network[0]=network
ucitrack.@network[0].init='network'
ucitrack.@network[0].affects='dhcp' 'radvd'
ucitrack.@wireless[0]=wireless
ucitrack.@wireless[0].affects='network'
ucitrack.@firewall[0]=firewall
ucitrack.@firewall[0].init='firewall'
ucitrack.@firewall[0].affects='luci-splash' 'qos' 'miniupnpd' 'sqm'
ucitrack.@olsr[0]=olsr
ucitrack.@olsr[0].init='olsrd'
ucitrack.@dhcp[0]=dhcp
ucitrack.@dhcp[0].init='dnsmasq'
ucitrack.@dhcp[0].affects='odhcpd'
ucitrack.@odhcpd[0]=odhcpd
ucitrack.@odhcpd[0].init='odhcpd'
ucitrack.@dropbear[0]=dropbear
ucitrack.@dropbear[0].init='dropbear'
ucitrack.@httpd[0]=httpd
ucitrack.@httpd[0].init='httpd'
ucitrack.@fstab[0]=fstab
ucitrack.@fstab[0].exec='/sbin/block mount'
ucitrack.@qos[0]=qos
ucitrack.@qos[0].init='qos'
ucitrack.@system[0]=system
ucitrack.@system[0].init='led'
ucitrack.@system[0].exec='/etc/init.d/log reload'
ucitrack.@system[0].affects='luci_statistics' 'dhcp'
ucitrack.@luci_splash[0]=luci_splash
ucitrack.@luci_splash[0].init='luci_splash'
ucitrack.@ntpclient[0]=ntpclient
ucitrack.@ntpclient[0].init='ntpclient'
ucitrack.@samba[0]=samba
ucitrack.@samba[0].init='samba'
ucitrack.@tinyproxy[0]=tinyproxy
ucitrack.@tinyproxy[0].init='tinyproxy'
ucitrack.@AdGuardHome[0]=AdGuardHome
ucitrack.@AdGuardHome[0].init='AdGuardHome'
ucitrack.@aria2[0]=aria2
ucitrack.@aria2[0].init='aria2'
ucitrack.@hd-idle[0]=hd-idle
ucitrack.@hd-idle[0].init='hd-idle'
ucitrack.@kcptun[0]=kcptun
ucitrack.@kcptun[0].init='kcptun'
ucitrack.minidlna=minidlna
ucitrack.@upnpd[0]=upnpd
ucitrack.@upnpd[0].init='miniupnpd'
ucitrack.@nlbwmon[0]=nlbwmon
ucitrack.@nlbwmon[0].init='nlbwmon'
ucitrack.@udpxy[0]=udpxy
ucitrack.@udpxy[0].init='udpxy'
ucitrack.@watchcat[0]=watchcat
ucitrack.@watchcat[0].init='watchcat'
ucitrack.@zerotier[0]=zerotier
ucitrack.@zerotier[0].init='zerotier'
ucitrack.@smartdns[0]=smartdns
ucitrack.@smartdns[0].init='smartdns'
ucitrack.@mwan3[0]=mwan3
ucitrack.@mwan3[0].exec='/etc/init.d/mwan3 reload'
ucitrack.@unbound[0]=unbound
ucitrack.@unbound[0].init='unbound'
ucitrack.@adbyby[0]=adbyby
ucitrack.@adbyby[0].init='adbyby'
ucitrack.@cifsd[0]=cifsd
ucitrack.@cifsd[0].init='cifsd'
ucitrack.@flowoffload[0]=flowoffload
ucitrack.@flowoffload[0].init='flowoffload'
ucitrack.@k3screenctrl[0]=k3screenctrl
ucitrack.@k3screenctrl[0].init='k3screenctrl'
ucitrack.@k3usb[0]=k3usb
ucitrack.@k3usb[0].init='k3usb'
ucitrack.@aliddns[0]=aliddns
ucitrack.@aliddns[0].init='aliddns'
ucitrack.@amule[0]=amule
ucitrack.@amule[0].init='amule'
ucitrack.@guest-wifi[0]=guest-wifi
ucitrack.@guest-wifi[0].init='guest-wifi'
ucitrack.@passwall[0]=passwall
ucitrack.@passwall[0].init='passwall'
ucitrack.@pppoe-relay[0]=pppoe-relay
ucitrack.@pppoe-relay[0].init='pppoe-relay'
ucitrack.@v2ray_server[0]=v2ray_server
ucitrack.@v2ray_server[0].init='v2ray_server'
ucitrack.@vlmcsd[0]=vlmcsd
ucitrack.@vlmcsd[0].init='kms'
ucitrack.@arpbind[0]=arpbind
ucitrack.@arpbind[0].init='arpbind'
ucitrack.@baidupcs-web[0]=baidupcs-web
ucitrack.@baidupcs-web[0].init='baidupcs-web'
ucitrack.@dnspod[0]=dnspod
ucitrack.@dnspod[0].init='dnspod'
ucitrack.@filebrowser[0]=filebrowser
ucitrack.@filebrowser[0].init='filebrowser'
ucitrack.@frp[0]=frp
ucitrack.@frp[0].init='frp'
ucitrack.@koolproxy[0]=koolproxy
ucitrack.@koolproxy[0].init='koolproxy'
ucitrack.@n2n_v2[0]=n2n_v2
ucitrack.@n2n_v2[0].init='n2n_v2'
ucitrack.@nps[0]=nps
ucitrack.@nps[0].init='nps'
ucitrack.@sqm[0]=sqm
ucitrack.@sqm[0].init='sqm'
ucitrack.@shadowsocksr[0]=shadowsocksr
ucitrack.@shadowsocksr[0].init='shadowsocksr'
ucitrack.@syncdial[0]=syncdial
ucitrack.@syncdial[0].exec='/bin/genwancfg'
ucitrack.@syncthing[0]=syncthing
ucitrack.@syncthing[0].exec='/etc/init.d/syncthing restart'
ucitrack.@vsftpd[0]=vsftpd
ucitrack.@vsftpd[0].init='vsftpd'
ucitrack.@xlnetacc[0]=xlnetacc
ucitrack.@xlnetacc[0].init='xlnetacc'
ucitrack.@mia[0]=mia
ucitrack.@mia[0].init='mia'
ucitrack.@mwan3helper[0]=mwan3helper
ucitrack.@mwan3helper[0].init='mwan3helper'
ucitrack.@pptpd[0]=pptpd
ucitrack.@pptpd[0].init='pptpd'
ucitrack.@qbittorrent[0]=qbittorrent
ucitrack.@qbittorrent[0].init='qbittorrent'
ucitrack.@ssocks[0]=ssocks
ucitrack.@ssocks[0].init='ssocks'

rufengsuixing, 2019-12-19:
ucitrack没问题，这样点击了保存应用会启动的啊，日志为空的话
请设置为/tmp/adguardhome.log

PoiYuudachi, 2019-12-19:
root@OpenWrt:~# uci show ucitrack
ucitrack.@network[0]=network
ucitrack.@network[0].init='network'
ucitrack.@network[0].affects='dhcp' 'radvd'
ucitrack.@wireless[0]=wireless
ucitrack.@wireless[0].affects='network'
ucitrack.@firewall[0]=firewall
ucitrack.@firewall[0].init='firewall'
ucitrack.@firewall[0].affects='luci-splash' 'qos' 'miniupnpd'
ucitrack.@olsr[0]=olsr
ucitrack.@olsr[0].init='olsrd'
ucitrack.@dhcp[0]=dhcp
ucitrack.@dhcp[0].init='dnsmasq'
ucitrack.@dhcp[0].affects='odhcpd'
ucitrack.@odhcpd[0]=odhcpd
ucitrack.@odhcpd[0].init='odhcpd'
ucitrack.@dropbear[0]=dropbear
ucitrack.@dropbear[0].init='dropbear'
ucitrack.@httpd[0]=httpd
ucitrack.@httpd[0].init='httpd'
ucitrack.@fstab[0]=fstab
ucitrack.@fstab[0].init='fstab'
ucitrack.@qos[0]=qos
ucitrack.@qos[0].init='qos'
ucitrack.@system[0]=system
ucitrack.@system[0].init='led'
ucitrack.@system[0].affects='luci_statistics' 'dhcp'
ucitrack.@cron[0]=cron
ucitrack.@cron[0].init='cron'
ucitrack.@luci_splash[0]=luci_splash
ucitrack.@luci_splash[0].init='luci_splash'
ucitrack.@ntpclient[0]=ntpclient
ucitrack.@ntpclient[0].init='ntpclient'
ucitrack.@samba[0]=samba
ucitrack.@samba[0].init='samba'
ucitrack.@tinyproxy[0]=tinyproxy
ucitrack.@tinyproxy[0].init='tinyproxy'
ucitrack.@koolddns[0]=koolddns
ucitrack.@koolddns[0].init='koolddns'
ucitrack.@AdGuardHome[0]=AdGuardHome
ucitrack.@AdGuardHome[0].init='AdGuardHome'
ucitrack.@aria2[0]=aria2
ucitrack.@aria2[0].init='aria2'
ucitrack.@upnpd[0]=upnpd
ucitrack.@upnpd[0].init='miniupnpd'
ucitrack.@nlbwmon[0]=nlbwmon
ucitrack.@nlbwmon[0].init='nlbwmon'
ucitrack.@flowoffload[0]=flowoffload
ucitrack.@flowoffload[0].init='flowoffload'
ucitrack.@kodexplorer[0]=kodexplorer
ucitrack.@kodexplorer[0].init='kodexplorer'
ucitrack.@arpbind[0]=arpbind
ucitrack.@arpbind[0].init='arpbind'
ucitrack.@shadowsocksr[0]=shadowsocksr
ucitrack.@shadowsocksr[0].init='shadowsocksr'
ucitrack.@vsftpd[0]=vsftpd
ucitrack.@vsftpd[0].init='vsftpd'
ucitrack.@unblockmusic[0]=unblockmusic
ucitrack.@unblockmusic[0].init='unblockmusic'

PoiYuudachi, 2019-12-19:
> ucitrack没问题，这样点击了保存应用会启动的啊，日志为空的话
> 请设置为/tmp/adguardhome.log

运行日志目录是这个 但日志就是空的

zxlhhyccc, 2019-12-19:
> 命令行输入即可

经测试，将base.lua替换为1.7.22版本可以启动，问题出在base.lua上面。。。。

zxlhhyccc, 2019-12-19:
> 尝试重装了一次，还是未能重现，无法提交页面有提示么，上面或者下面

没有提示，有提示早就提交了。。。。

zxlhhyccc, 2019-12-19:
> 尝试重装了一次，还是未能重现，无法提交页面有提示么，上面或者下面

卸载旧版本重新安装新版本，好像没事，如果是新编译进去的就是这个问题。。。。

zxlhhyccc, 2019-12-19:
> 尝试重装了一次，还是未能重现，无法提交页面有提示么，上面或者下面

经测试，一直到1.7.27版本没问题，从1.7.28版本开始就出现这问题。。。。

rufengsuixing, 2019-12-19:
27-28我改变了提交方式为ucitrack，如果ucitrack失效的话很可能你的设备上的提交会失效，具体为什么ucitrack会失效我也不知道

rufengsuixing, 2019-12-19:
如果ucitrack失效其他应用的开关应该也会受影响

rufengsuixing, 2019-12-19:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/dcfbe0703fdd0dce6bb56a7e25372f4fdea06ec3

zxlhhyccc, 2019-12-19:
> [dcfbe07](https://github.com/rufengsuixing/luci-app-adguardhome/commit/dcfbe0703fdd0dce6bb56a7e25372f4fdea06ec3)

这个commit是修改的？

rufengsuixing, 2019-12-19:
不是，是以前的，
涉及到这个问题的修改

rufengsuixing, 2019-12-19:
测试一下
lock -n /var/run/luci-reload

zxlhhyccc, 2019-12-19:
> 测试一下
> lock -n /var/run/luci-reload

没反应。。。。
[image]

rufengsuixing, 2019-12-19:
问题没找到，但是我可以出一个临时修复方案，只是重启后第一次进界面需要点击两次才会生效

zxlhhyccc, 2019-12-19:
> 问题没找到，但是我可以出一个临时修复方案，只是重启后第一次进界面需要点击两次才会生效
点击没运行的情况下的：
[image]

zxlhhyccc, 2019-12-19:
> 问题没找到，但是我可以出一个临时修复方案，只是重启后第一次进界面需要点击两次才会生效

我暂时退回1.7.27版本。。。。

rufengsuixing, 2019-12-19:
能够确认的是你的设备上的ucitrack 损坏了，我刚才尝试创造了个循环的启动项，同样导致了我的设备上的ucitrack损坏，并且我除了重置之外没有办法修复

rufengsuixing, 2019-12-19:
在无法载入时给luci解锁
lock -u /var/run/luci-reload 
就发现开关按照想法进行了变动，
至于哪里出了问题，可能是uci在异常终止的时候的bug，这个超出了本luci的能力范围

closed 2019-12-19 by rufengsuixing

zxlhhyccc, 2019-12-19:
> 能够确认的是你的设备上的ucitrack 损坏了，我刚才尝试创造了个循环的启动项，同样导致了我的设备上的ucitrack损坏，并且我除了重置之外没有办法修复

能否不使用ucitrack？

zxlhhyccc, 2019-12-19:
> 问题没找到，但是我可以出一个临时修复方案，只是重启后第一次进界面需要点击两次才会生效

刚编译了你的新源码，有点好转，不过在不加入gfw列表情况下需要点三次保存并应用才能运行。。。。

rufengsuixing, 2019-12-20:
ucitrack才是标准的启动模式，其他的都是各种hack而已，我在最新的里面加入了检测到连续两次提交便不使用ucitrack的调整

zxlhhyccc, 2019-12-20:
> ucitrack才是标准的启动模式，其他的都是各种hack而已，我在最新的里面加入了检测提交失效，可以提交两次便可以检测ucitrack是否正常，调整是否使用ucitrack

能不能改为提交一次即可？就是说自动检测正常情况下就执行运行。。。

rufengsuixing, 2019-12-20:
提交两次这种检测只需要一次即可，不会每次都要反复提交的

zxlhhyccc, 2019-12-20:
> 提交两次这种检测只需要一次即可，不会每次都要反复提交的

你的意思是说第一次需要检测而两次提交，以后即使停止再重新提交一次就可以运行？