rufengsuixing/luci-app-adguardhome #46: 潘多拉固件不能更新核心bin

qpzr, 2019-12-23:
是否支持3.14内核的潘多拉固件？
每次安装后更新主程序bin
都是英文的 更新失败 请稍后再试

rufengsuixing, 2019-12-23:
请按照模板贴上日志

liyiecho, 2019-12-24:
```
 Local version: v0.100.6., cloud version: v0.100.7.
Updating core...
start download
--2019-12-24 19:20:29--  https://github.com/AdguardTea
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-12-24 19:20:31 ERROR 404: Not Found.

download failed try another download
--2019-12-24 19:20:31--  http://m/AdGuardHome/releases/dow
Resolving m... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'm'
download failed try another download
--2019-12-24 19:20:31--  http://nload/v0.100.7/AdGuardHo
Resolving nload... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'nload'
download failed try another download
--2019-12-24 19:20:31--  http://me_linux_amd64.tar.gz/
Resolving me_linux_amd64.tar.gz... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'me_linux_amd64.tar.gz'
download failed try another download
--2019-12-24 19:20:31--  https://static.adguard.com/adg
Resolving static.adguard.com... 89.187.185.162
Connecting to static.adguard.com|89.187.185.162|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-12-24 19:20:31 ERROR 404: Not Found.

download failed try another download
--2019-12-24 19:20:31--  http://uardhome/release/AdGuardHo
Resolving uardhome... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'uardhome'
download failed try another download
--2019-12-24 19:20:32--  http://me_linux_amd64.tar.gz/
Resolving me_linux_amd64.tar.gz... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'me_linux_amd64.tar.gz'
download failed try another download
--2019-12-24 19:20:32--  http://guardhome/beta/AdGuardHom
Resolving guardhome... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'guardhome'
download failed try another download
--2019-12-24 19:20:32--  http://e_linux_amd64.tar.gz/
Resolving e_linux_amd64.tar.gz... failed: Name does not resolve.
wget-ssl: unable to resolve host address 'e_linux_amd64.tar.gz'
download failed try another download
no download success
```

看样子是把升级用的下载链接分行读取成下面这样了，

```
> https://github.com/AdguardTea
> m/AdGuardHome/releases/dow
> nload/${latest_ver}/AdGuardHo
> me_linux_${Arch}.tar.gz
> https://static.adguard.com/adg
> uardhome/release/AdGuardHo
> me_linux_${Arch}.tar.gz
> #https://static.adguard.com/ad
> guardhome/beta/AdGuardHom
> e_linux_${Arch}.tar.gz
```

原本应该是这样的

```
> https://github.com/AdguardTeam/AdGuardHome/releases/download/${latest_ver}/AdGuardHome_linux_${Arch}.tar.gz
> https://static.adguard.com/adguardhome/release/AdGuardHome_linux_${Arch}.tar.gz
> #https://static.adguard.com/adguardhome/beta/AdGuardHome_linux_${Arch}.tar.gz
```

rufengsuixing, 2019-12-24:
luci链接修改后提交呢

rufengsuixing, 2019-12-24:
我的默认链接行数是对的，是你后来修改的？

rufengsuixing, 2019-12-24:
请提交你用的浏览器是什么，以及是否可以在luci界面修改链接恢复正常工作

liyiecho, 2019-12-24:
@rufengsuixing 无论luci界面或者后台修改链接为
```
https://github.com/AdguardTeam/AdGuardHome/releases/download/${latest_ver}/AdGuardHome_linux_${Arch}.tar.gz
https://static.adguard.com/adguardhome/release/AdGuardHome_linux_${Arch}.tar.gz
#https://static.adguard.com/adguardhome/beta/AdGuardHome_linux_${Arch}.tar.gz
```
都会被系统分行。且分行的长度跟luci那的长度一致
[image]

我用的是chrome

rufengsuixing, 2019-12-24:
warp on的默认值应该是不提交自动回行产生的空格的，估计是你的浏览器问题，请提交浏览器型号

rufengsuixing, 2019-12-24:
我会用chrome重新确认一下，稍等

liyiecho, 2019-12-24:
Version 79.0.3945.88 (Official Build) (64-bit)
[image]

rufengsuixing, 2019-12-24:
问题已确认已修复，可能只在chrome上存在

rufengsuixing, 2019-12-24:
release 已更新修复

closed 2019-12-24 by rufengsuixing