rufengsuixing/luci-app-adguardhome #48: 第一次启动程序无法自动下载主程序

yangsongli, 2019-12-26:
AdGuard Home luci的版本:1.8-5
操作系统和版本:openwrt R9.12.03 x86架构
希望的执行结果
希望第一次启用程序后自动下载AdGuard Home主程序
实际的执行结果
第一次启动然后一直提示A task is already running. 就没有任何反应 必须要重启openwrt才会自动下载AdGuard Home主程序
日志
/tmp/AdGuardHome_update.log的日志只有A task is already running.
INIT_TRACE=1 /etc/init.d/AdGuardhome reload 2>/tmp/debug 后的/tmp/debug文件的日志只有-ash: /etc/init.d/AdGuardhome: not found
截图
[image]

yangsongli, 2019-12-26:
已解决 单臂openwrt才会出现这种情况 有wan口就不会

rufengsuixing, 2019-12-26:
A task is already running表示已经在下载了，最好不要打断下载进程

rufengsuixing, 2019-12-26:
not found 表示你并没有成功安装

rufengsuixing, 2019-12-27:
我会尝试优化正在更新时再次提交更新核心时的显示，尽量显示正在下载的日志而不是显示正在下载

rufengsuixing, 2019-12-27:
https://github.com/rufengsuixing/luci-app-adguardhome/commit/5eecff1964885918504a156d6d7fe0db72a1221d

closed 2019-12-27 by rufengsuixing