rufengsuixing/luci-app-adguardhome #49: luci界面上关闭后，重新启用需要点两次保存并应用才能启用

liyiecho, 2019-12-27:
很久没关闭过，但最近发现luci界面上关闭后，重新启用需要点两次保存并应用才能启用。
目前版本1.8-7依然有这个问题。

点第一次保存并应用是这样的：
[image]
[image]

点第二次保存并应用就正常启用了。
[image]
[image]

rufengsuixing, 2019-12-27:
你的 ucitracktest为零，证明你的ucitrack至少在我这个luci这里是无效的，我这里的解决方法是第一次进行检测，当然第一次的提交是失效的，之后就会切换提交方式，就会正常

你可以测试一下是不是reload时机早于了提交配置，
具体来说就是拨动启用开关并提交保存是不是运行状态相反

liyiecho, 2019-12-27:
固件版本	OpenWrt SNAPSHOT r11816-d5ede68f8b / LuCI Master git-19.360.66168-1ccb8e1
内核版本	4.19.91
[image]

luci界面上关闭应用，刚开始依然显示运行，此时后台查看AdGuardHome已经结束运行了，等待几秒后，luci界面才显示停止

rufengsuixing, 2019-12-27:
运行状态同步总是会有延迟的，这个是正常的

rufengsuixing, 2020-01-03:
close due to inactive

closed 2020-01-03 by rufengsuixing