rufengsuixing/luci-app-adguardhome #50: 原版openwrt-19.07关闭后无法在后台启动.

garypang13, 2019-12-28:
* **AdGuard Home luci的版本:**
1.8-8

* **操作系统和版本:**
 OpenWrt / LuCI openwrt-19.07 branch git-19.356.80404-43c74d3

使用的最新原版openwrt 19.07,可以在后台关闭adg,但是关闭后无法再开启,关闭后也不再有任何日志,
在SSH使用命令
uci set AdGuardHome.AdGuardHome.enabled='1'
uci commit AdGuardHome
也无效
但是使用 service AdGuardHome start 可以正常开启.

rufengsuixing, 2019-12-28:
没有19.07，没有测试环境，无能为力

rufengsuixing, 2019-12-28:
开启一般是/etc/init.d/AdGuardHome reload

liyiecho, 2019-12-28:
[image]

`uci commit`只是保存修改，并没有其他能力的

[image]

具体的可以看
https://openwrt.org/docs/guide-user/base-system/uci

Mosney, 2019-12-30:
保存配置文件跟运行有什么关系？这个issue可以直接关闭了

garypang13, 2019-12-30:
@rufengsuixing 方便的话可以下载我编译的原版带这个插件的openwrt-19.07 X86_64固件测试下,毕竟除了国内小部分,国外都是用的原版op,下载地址:https://cowtransfer.com/s/75ad7c3d0ba54d

closed 2020-01-03 by rufengsuixing