rufengsuixing/luci-app-adguardhome #51: 敢问Adh是怎么同ssr-plus合作的

doumeki, 2020-01-03:
dns转发那里只填了adh的端口, adh设置为上游解析,也只填当地的dns. 我看了下日志, 把youtube返回127.1,但是能访问,肯定是ssr-plus里的pdnsd解析了, 敢问他们是怎么合作的,adh有将gfw的转发给pdnsd了?

rufengsuixing, 2020-01-03:
luci设置项里面有上游，默认是直接发给opendns，解析是没问题的也不会发给pdnsd，但是如果用gfw列表的话，ipset功能缺失

rufengsuixing, 2020-01-03:
这样的话，一旦adh炸鸡了，恢复了默认dnsmasq不会导致dns无法解析

doumeki, 2020-01-03:
感谢.我看了下我的LUCI, 是没有添加更多选项时的内容(就是GFW增加删除什么的), 基本上是默认设置. 
说明应该默认就是转发现到ssr-plus里的pdnsd (5335端口)?
[image]

rufengsuixing, 2020-01-03:
如果使用dnsmasq作为上游服务器，相当于串联，就是adh收到的请求会发到dnsmasq那里，gfwlist的处理和adh没关系了

rufengsuixing, 2020-01-03:
还有为啥你的界面是英文，我的po翻译失效了么

doumeki, 2020-01-03:
好的,感谢 虽然我没完全明白,下来我研究下.
我以前安过你的luci是中文的,这个是别人编译的,我没升级你最新的所以是英语 :)

rufengsuixing, 2020-01-03:
你的设置是
你--dnsmasq--（adh或者pdnsd）

doumeki, 2020-01-03:
我倒是明白我设置的,从client (有redirect所有53 "iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53")-->dnsmasq, --> adh. 我想的是默认设置下是不是有什么地方配置的有gfw列表就不走adh而走其它的,比如pdnsd.
但是我始终没有找到adh是在哪使用的gfw列表的. 这个项目是老外的吧,理论上说不会专门还搞个gfw列表.
[image]
这个我的dns转发设置, 只有adh,也没转发到pdnsd上去.所以我一直没明白是在哪分流的?

rufengsuixing, 2020-01-03:
分流是dnsmasq做的，可以ps www | grep dnsmasq 从命令行中看一下配置文件
发现是在/tmp/dnsmasq.ssr中写的列表

doumeki, 2020-01-03:
好的.感谢. 终于找到了.

closed 2020-01-03 by rufengsuixing