rufengsuixing/luci-app-adguardhome #52: 使用53重定向到Adghome，dhcp失效，Adghome自带dhcp不生效。

Kaze1027, 2020-01-08:
### 问题背景情况
- [Q ] 我用着最新版的 luci-app-adguardhome
- [A ] 是的。
- [Q ] 我检查了没有人提交同样的问题
- [A ] 仅有一人提问53替代dnsmasq失败，可能是同样的原因。
- [Q ] 我所使用的openwrt版本
- [A ] eSir编译的ipv4版本19年12月版本，基于lean源码。链接：https://drive.google.com/open?id=1p097_lAvo8OiRkTlDw_bkFvRs69Eckxm
- [Q ] 我使用的 luci-app-adguardhome大致配置
- [A ] 端口为7913，将7913重定向到53。
- [Q ] Adghome工作状态如何
- [A ] 在设备获取有效ip时，可以正常解析。详情：在刚开机的几个小时内能够正常使用，经过几个小时之后便无法使用。原因可能是因为我启用了Adghome延迟启动，刚开始设备通过dnsmasq正常获取到了ip，当adghome启动后，设备拥有正常的ip，可以正常解析。但当ip租期过了之后，没有dhcp服务导致设备无法正常上网。
- [Q ] 其他补充
- [A ] lean源码的都有两条自定义防火墙规则：
iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53
-------

### 问题详细信息
lean源码的固件，在启用端口重定向到53后，再注释掉lean源码自定义防火墙规则“iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53”即可正常工作，但是在下属设备ip租期过后，设备无法获得新的ip或是续租ip。

* **AdGuard Home luci的版本:**
  * <1.8-11>
* **AdGuard Home server的版本:**
  * <v0.99.2>
* **操作系统和版本:**
  * <具体版本不清楚，较新>
-------

### 希望的执行结果
<希望adguardhome正常运行即可>
### 实际的执行结果
<adguardhome能够正常作为本地dns解析服务器，但会使得OpenWrt的dhcp失效>

-------

### 其他情况
<eSir的固件中的科学￥上网插件会在保存后清空防火墙再启动，这样使得科学￥上网插件的dns自定义解析生效了，这个情况可作参考。>

------

### 截图
[image]
[image]
[image]

rufengsuixing, 2020-01-08:
请升级AdGuard Home server的版本

rufengsuixing, 2020-01-08:
本luci没有提供dhcp服务，另外通过ps |grep dnsmasq检测dhcp服务是否正常运行

rufengsuixing, 2020-01-09:
确认在部分设备上出现dns失败的问题，然而现在仍然没有找到原因

rufengsuixing, 2020-01-09:
破案了，和本软件无关
可以在接口高级设置中
[image]
设定本机dns，至于为什么不正常，暂时不知道，只知道为空的话
获取到的默认dns为
                                        fec0:0:0:ffff::1%1
                                       fec0:0:0:ffff::2%1
                                       fec0:0:0:ffff::3%1
估计与残留的ipv6接口有关

rufengsuixing, 2020-01-09:
不过也给我提供了一个新思路，可以通过让ipv6机器获取到ipv4的dns这样来避免ipv6机器地址统计困难

Kaze1027, 2020-01-09:
好的，非常感谢，我去试一试目前一大问题就是重定向后dnsmasq无法正常下发ip，原因不明

rufengsuixing, 2020-01-09:
windows尝试ipv6dns如果失败之后不会回落到ipv4，所以一切干扰ipv6 dns的手段都会导致windows找不到网页，如果不指定ipv6
dns的话windows就会用那三个连不上的作为替代，23333

closed 2020-01-24 by Kaze1027

Phillweston, 2022-12-15:
> 破案了，和本软件无关 可以在接口高级设置中 [image] 设定dhcp 下发的dns，至于为什么不正常，暂时不知道，只知道为空的话 获取到的默认dns为 fec0:0:0:ffff::1%1 fec0:0:0:ffff::2%1 fec0:0:0:ffff::3%1 就是windows自行用的默认dns地址，连不上就找不到网页 有的固件不会自动设置路由本机为dns服务器就会造成这样的问题

我参照此方法进行修改了，目前运行正常，等过几天再有同样的问题再来反馈。

今天遇到的同样问题了，我在此issue下面补充点log，供后人参考
我的固件是**Lean iStore**版本，配置是**smartdns+adguard home+passwall**

passwall没有用那个xray，可能会导致断流，参考此issue问题
https://github.com/xiaorouji/openwrt-passwall/issues/1431

smartdns设置为dnsmasq的上游服务器，adguard53重定向至1745
遇到的问题也是断流然后smartdns提示重定向到dnsmasq失败。此时只能手动重启smartdns服务然后在DHCP/DNS服务中重新填写DNS转发然后保存才能正常使用。

补充一下终端输出的log日志，大家也可以帮我分析一下问题跟之前的各位是否一致：
```
root@OpenWrt:/tmp/log# cat passwall.log
2022-12-15 20:33:33: 执行启动延时 1 秒后再启动!
2022-12-15 20:33:35: TCP节点：[🇭🇰 香港 01]QmZPuqxWpD7UPmAJDiTY.ganode.org:1032，监听端口：1041
2022-12-15 20:33:35: 过滤服务配置：准备接管域名解析...
2022-12-15 20:33:35:   - 域名解析：使用UDP协议请求DNS（127.0.0.1#6553）...
2022-12-15 20:33:35:   * 要求代理 DNS 请求，如上游 DNS 非直连地址，确保 UDP 代理打开，并且已经正确转发！
2022-12-15 20:33:35:   - 节点列表中的域名(vpsiplist)使用分组：cn
2022-12-15 20:33:35:   - 域名白名单(whitelist)使用分组：cn
2022-12-15 20:33:35:   - 节点订阅域名(blacklist)使用分组：passwall_proxy
2022-12-15 20:33:35:   - 代理域名表(blacklist)使用分组：passwall_proxy
2022-12-15 20:33:36:   - 防火墙域名表(gfwlist)使用分组：passwall_proxy
2022-12-15 20:33:36:   - 中国域名表(chnroute)使用分组：cn
2022-12-15 20:33:36:   - 请让SmartDNS作为Dnsmasq的上游或重定向！
2022-12-15 20:33:40:   - 域名解析：使用SmartDNS，请确保配置正常。
2022-12-15 20:33:40: 开始加载防火墙规则...
2022-12-15 20:33:41: 加入负载均衡的节点到ipset[vpsiplist]直连完成
2022-12-15 20:33:41: 加入所有节点到ipset[vpsiplist]直连完成
2022-12-15 20:33:42: 加载路由器自身 TCP 代理...
2022-12-15 20:33:42:   - [0]，屏蔽代理UDP 端口：80,443
2022-12-15 20:33:42: 加载路由器自身 UDP 代理...
2022-12-15 20:33:42:   - 上游 DNS 服务器 127.0.0.1 已在直接访问的列表中，不强制向 UDP 代理转发对该服务器 UDP/6553 端口的访问
2022-12-15 20:33:42: TCP默认代理：使用TCP节点[🇭🇰 香港 01] [防火墙列表](REDIRECT:1041)代理所有端口
2022-12-15 20:33:43: UDP默认代理：使用UDP节点[🇭🇰 香港 01] [防火墙列表](TPROXY:1041)代理所有端口
2022-12-15 20:33:43: 防火墙规则加载完成！
2022-12-15 20:33:47: 重启 dnsmasq 服务
2022-12-15 20:33:47: 配置定时任务：自动更新【GaCloud】订阅。
2022-12-15 20:33:47: 运行完成！

root@OpenWrt:/tmp/log# cat smartdns.log 
[2022-12-15 20:33:43,256][ERROR][      fast_ping.c:462 ] sendto 2a01:111:2003::52, id 184, Permission denied
[2022-12-15 20:38:42,856][ERROR][      fast_ping.c:462 ] sendto 2a01:111:2003::52, id 989, Permission denied
[2022-12-15 21:04:25,585][ERROR][      fast_ping.c:462 ] sendto 2a02:6ea0:d600::2, id 134, Permission denied
```