rufengsuixing/luci-app-adguardhome #55: 更新地址失效了

czghzh, 2020-01-11:
### Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version of luci-app-adguardhome
- [ ] I checked to make sure that this issue has not already been filed

### Issue Details

* **Version of AdGuard Home luci:**
  * 
* **Version of AdGuard Home server:**
  * 
* **Operating system and version:**
  * 

### Expected Behavior

### Actual Behavior

### Screenshots

<details><summary>Screenshot:</summary>

</details>

### Additional Information

rufengsuixing, 2020-01-11:
sh -x /usr/share/AdGuardHome/update_core.sh 
完整日志贴上来

komanchun, 2020-02-12:
我也不能更新

else05, 2020-03-03:
```
Local version: ., cloud version: v0.100.9.
Updating core...
start download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   620  100   620    0     0     80      0  0:00:07  0:00:07 --:--:--   156
100 6258k  100 6258k    0     0   422k      0  0:00:14  0:00:14 --:--:-- 1577k
download success start copy
start copy
awk: /etc/AdGuardHome.yaml: No such file or directory
AdGuardHome service disabled
Command failed: Not found
rm: can't remove '/usr/bin/AdGuardHome/AdGuardHome': No such file or directory
awk: /etc/AdGuardHome.yaml: No such file or directory
Succeeded in updating core.
Local version: v0.100.9, cloud version: v0.100.9.
```

第一安装，脚本是不是要增强一下判断，谢谢

else05, 2020-03-03:
在界面上点手动设置 ，把里面的内容复制一下，
`vim /etc/AdGuardHome.yaml `
然后把内容粘上去保存，
再执行更新核心就可以了，我的是`L大`的固件，我自己折腾就是这么处理的