rufengsuixing/luci-app-adguardhome #57: 高恪固件，安装完后无法找到入口

simakuang, 2020-01-20:
实在抱歉，第一次在Github上提问，自己对于路由器也是一知半解，如果有描述的不清楚或者问题太脑残可以忽视我：）

依葫芦画瓢网上搜寻资料后，成功用opkg命令安装了ipk，但是并不能在目录中找到adguard的入口。
用同样方法安装了网上搜寻的其他ipk，（如koolproxy），会在网络设置中看到入口。

opkg的源已经更新，且安装过程未报错。

高恪版本：稳定分支 4.2.7.17422

无论如何，十分感谢大佬编译的luci版本~

simakuang, 2020-01-20:
[image]

simakuang, 2020-01-20:
[image]

rufengsuixing, 2020-01-22:
op的包，不是给高格使用的，高格我记得是个非常老的op版本，luci版本未知，看你提供的信息，可能是在高格中没有service这个列表，你可以尝试修改下index 在/usr/lib/lua/luci/controller/AdGuardHome.lua中 ,尝试将service修改为network看是否能显示出界面

simakuang, 2020-01-30:
试了一下不行，还是非常感谢。打算刷个openwrt试一下啦~

closed 2020-01-30 by simakuang