rufengsuixing/luci-app-adguardhome #58: ADH和SSR搭配的设置问题

poppyseven, 2020-02-06:
- [*] 我用着最新版的 luci-app-adguardhome
- [*] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  * 1.8-11
* **操作系统和版本:**
  *  OpenWrt R20.1.24

### 问题
ADH的监听端口设置为"5335"
重定向模式为“重定向53端口到ADH”
SSR运行模式为“绕过大陆IP”
SSR的DNS的解析方式设置为“使用本机端口为5335的DNS服务”

目前上述设置上网正常，想请教一下这样设置是否让ADH起到了作用？
另外在ADH的上游服务器里添加了国内和国外的DNS服务器，是否会有冲突？

### 截图

<details><summary>截图:</summary>

[image]

</details>

### 更多的信息

poppyseven, 2020-02-06:
补充一下，有上述疑惑主要是因为用了ADH之后，tw.manhuagui.com这个网站无法打开，ping不同。但在没有使用ADH之前该网站浏览正常。

XeonLang, 2020-02-24:
重定向模式设为“使用53端口替换dnsmasq”一切正常