rufengsuixing/luci-app-adguardhome #63: 无法下载内核文件

frank2002, 2020-03-02:
- [Y] I am running the latest version of luci-app-adguardhome
- [Y] I checked to make sure that this issue has not already been filed

 OpenWrt SNAPSHOT, r2112-15b0fffb

安装过程中显示
Configuring luci-app-adguardhome.
uci: Entry not found
uci: Entry not found
Command failed: Not found
awk: /etc/AdGuardHome.yaml: No such file or directory
Command failed: Invalid argument

安装完成后点击更新核心版本，提示：A task is already running. 内核没有下载下来，日志是空白的，没有报错。
当将各个路径放到/tmp下面之后下载开始正常，提示有内核文件，并且可以访问web面板，但是重启以后就消失了。

本人使用的是通过Lean源码编译的OpenWrt最新版，在vmware下作为旁路由使用。
[image]
[image]
[image]

else05, 2020-03-03:
在界面上点快速配置， ，把里面的内容复制一下，ssh连接到路由器中，
`vim /etc/AdGuardHome.yaml `
然后把内容粘上去保存，
开启全局代理上网，再执行更新核心就可以了，我的是`L大`的固件，我自己折腾就是这么处理的

frank2002, 2020-03-03:
> 在界面上点快速配置， ，把里面的内容复制一下，ssh连接到路由器中，
> `vim /etc/AdGuardHome.yaml `
> 然后把内容粘上去保存，
> 开启全局代理上网，再执行更新核心就可以了，我的是`L大`的固件，我自己折腾就是这么处理的

感谢你的回复，
我试了你的方法，但是提示框里依然是: a task is already running.
但是显示有配置文件了
[image]
（没有日志）

else05, 2020-03-13:
> > 在界面上点快速配置， ，把里面的内容复制一下，ssh连接到路由器中，
> > `vim /etc/AdGuardHome.yaml `
> > 然后把内容粘上去保存，
> > 开启全局代理上网，再执行更新核心就可以了，我的是`L大`的固件，我自己折腾就是这么处理的
> 
> 感谢你的回复，
> 我试了你的方法，但是提示框里依然是: a task is already running.
> 但是显示有配置文件了
> [image]
> （没有日志）

你要挂代理哦

else05, 2020-03-13:
`sh -x /usr/share/AdGuardHome/update_core.sh`  执行看看有什么报错呢

frank2002, 2020-03-13:
[image]

else05, 2020-03-14:
> [image]
> 代理用的就是大雕的ssrp

重新执行一下呢，不要`-x` 参数, ` sh  /usr/share/AdGuardHome/update_core.sh`
如果正常情况是会出现如下的下载进度，如果没有，使用`ps -ef |grep update_core`查看进程id, kill掉，然后再执行上面命令进行更新，我都是这样更新的
```shell
root@OpenWrt:~# sh  /usr/share/AdGuardHome/update_core.sh
/usr/bin/curl
param : curl -L -k --retry 2 --connect-timeout 20 -o - https://api.github.com/repos/AdguardTeam/AdGuardHome/releases/latest 2>/dev/null|grep -E 'tag_name' |grep -E 'v[0-9.]+' -o 2>/dev/null
Local version: v0.100.9., cloud version: v0.101.0.
Updating core...
start download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   620  100   620    0     0    447      0  0:00:01  0:00:01 --:--:--   447
 24 6413k   24 1563k    0     0  37641      0  0:02:54  0:00:42  0:02:12 14780
```

else05, 2020-03-14:
我手动更新成功的完整日志如下，`sh  /usr/share/AdGuardHome/update_core.sh`
```shell
root@OpenWrt:~# sh  /usr/share/AdGuardHome/update_core.sh
/usr/bin/curl
param : curl -L -k --retry 2 --connect-timeout 20 -o - https://api.github.com/repos/AdguardTeam/AdGuardHome/releases/latest 2>/dev/null|grep -E 'tag_name' |grep -E 'v[0-9.]+' -o 2>/dev/null
Local version: v0.100.9., cloud version: v0.101.0.
Updating core...
start download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   620  100   620    0     0    268      0  0:00:02  0:00:02 --:--:--   268
100 6413k  100 6413k    0     0  1134k      0  0:00:05  0:00:05 --:--:-- 2041k
download success start copy
start copy
udhcpc: started, v1.30.0
udhcpc: sending discover
udhcpc: no lease, failing
AdGuardHome service disabled
Command failed: Not found
workdir is a f2fs filesystem
udhcpc: started, v1.30.0
udhcpc: sending discover
udhcpc: no lease, failing
AdGuardHome service enabled
luci enable switch=1
Succeeded in updating core.
Local version: v0.101.0, cloud version: v0.101.0.
```

frank2002, 2020-03-18:
感谢，完美解决了！

closed 2020-03-18 by frank2002

neosoul9818, 2021-06-14:
> > 在界面上点快速配置， ，把里面的内容复制一下，ssh连接到路由器中，
> > `vim /etc/AdGuardHome.yaml `
> > 然后把内容粘上去保存，
> > 开启全局代理上网，再执行更新核心就可以了，我的是`L大`的固件，我自己折腾就是这么处理的
> 
> 感谢你的回复，
> 我试了你的方法，但是提示框里依然是: a task is already running.
> 但是显示有配置文件了
> [image]
> （没有日志）
哪里有快速配置？复制哪些内容？

qinghuas, 2022-12-18:
我的执行文件路径是
```
/usr/bin/AdGuardHome/AdGuardHome
```
但是 `cd /usr/bin/AdGuardHome` 后只有一个 `data` 文件夹

我是这么解决的（注意修改版本号和架构如 `amd64`）

如果路由器不能翻墙，可以先下载下来然后传上去

```
cd /usr/bin/AdGuardHome
wget https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.107.21/AdGuardHome_linux_amd64.tar.gz
tar -zxvf AdGuardHome_linux_amd64.tar.gz
mv AdGuardHome AdGuardHome_dir
mv AdGuardHome_dir/AdGuardHome ./
```
然后返回插件页面启动就可以用了