rufengsuixing/luci-app-adguardhome #68: 19.07.2 安装后无法启动

koakon, 2020-03-19:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [ ] 我用着最新版的 luci-app-adguardhome
- [ ] 我检查了没有人提交同样的问题

### 问题详细信息

Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: grep: /etc/crontabs/root: No such file or directory
Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: grep: /etc/crontabs/root: No such file or directory
Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: grep: /etc/crontabs/root: No such file or directory
Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: grep: /etc/crontabs/root: No such file or directory
Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: grep: /etc/crontabs/root: No such file or directory
Thu Mar 19 01:16:25 2020 daemon.err uhttpd[1754]: sh: write error: Broken pipe

linxuankent, 2020-03-19:
核心要怎么安装，我也只安装了管理界面

linxuankent, 2020-03-19:
sh -x /usr/share/AdGuardHome/update_core.sh 
ssh执行这个我就正常了