rufengsuixing/luci-app-adguardhome #70: Fails to install, fails to uninstall

kstenerud, 2020-03-26:
- [x] I am running the latest version of luci-app-adguardhome
- [x] I checked to make sure that this issue has not already been filed

### Issue Details

* **Version of AdGuard Home luci:**
  * v1.8-11
* **Version of AdGuard Home server:**
  * v0.101.0
* **Operating system and version:**
  * OpenWrt 19.07.2 r10947-65030d81f3

### Console log (installing)

```
root@OpenWrt:~# opkg install luci-app-adguardhome_1.8-11_all.ipk 
Installing luci-app-adguardhome (1.8-11) to root...
Configuring luci-app-adguardhome.
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
awk: /etc/AdGuardHome.yaml: No such file or directory
```

### Console log (uninstalling)

```
root@OpenWrt:~# opkg remove luci-app-adguardhome
Removing package luci-app-adguardhome from root...
awk: /etc/AdGuardHome.yaml: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
AdGuardHome service disabled
awk: /etc/AdGuardHome.yaml: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
AdGuardHome service disabled
Command failed: Not found
Not deleting modified conffile /etc/config/AdGuardHome.
```

py14551, 2020-04-01:
同样的问题 如何解决的？

py14551, 2020-04-01:
same problem