rufengsuixing/luci-app-adguardhome #79: passwall 的 gfwlist 模式无法与 ADH 配合使用

FishMagic, 2020-05-03:
### 问题详细信息
* **AdGuard Home luci的版本:**
  * 1.8-9
* **AdGuard Home server的版本:**
  * 0.101.0
* **操作系统和版本:**
  * OpenWrt R20.4.8

### 希望的执行结果
以一台树莓派作为旁路由，在其上运行 ADH 和 passwall，并以 gfwlist 模式为内网所有设备提供科学上网和去广告服务
### 实际的执行结果
去广告服务运行正常，但是 passwall 运行时却会出现异常：
  * ADH 为 dnsmasq 的上游时
    + 使用大陆白名单模式，百度和谷歌访问均正常
    + 使用 gfwlist 模式，百度访问正常，谷歌访问不正常
  * ADH 替换 dnsmasq 时，使用 gfwlist 模式或大陆白名单模式，均为百度访问正常，谷歌访问不正常
### 日志(重要)
[https://paste.ubuntu.com/p/zrHZC6Y64v/](https://paste.ubuntu.com/p/zrHZC6Y64v/)
### 截图
<details> <summary>截图:</summary>

[image]

[image]

[image]

[image]

</details>

### 更多的信息
ADH 作为 dnsmasq 上游时，DNS 请求返回的结果不是 127.0.0.1#5053 的解析结果，而是 192.168.61.1#53 即主路由的解析结果

zhea55, 2020-05-04:
@FishMagic  adguard 作为dnsmasq的上游的时候，其实dnsmasq默认又一个ipset，你可以前台运行dnsmasq，命中ipset里面的ip，请求就不会走到adguard，直接用ipset里面的值返回了。

FishMagic, 2020-05-04:
@zhea55 不知可否说明一下详细的操作方法。。

zhea55, 2020-05-04:
@FishMagic 
ps -ef | grep dnsmasq
执行这个得到下面这一行
 /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411 

在最后面加上--no-daemon --log-queries

回车以后，就能看到它每一个dns的请求。用nslookup google.com 就会发现它命中了ipset

FishMagic, 2020-05-04:
@zhea55 那么在我之前的配置上搭配使用 [autoipsetadder](https://github.com/rufengsuixing/luci-app-autoipsetadder) 是不是就可以让 ADH 作为 dnsmasq 的上游同时可以使用 gfwlist 进行科学上网

FishMagic, 2020-05-04:
> 那么在我之前的配置上搭配使用 [autoipsetadder](https://github.com/rufengsuixing/luci-app-autoipsetadder) 是不是就可以让 ADH 作为 dnsmasq 的上游同时可以使用 gfwlist 进行科学上网

成功解决问题，gfwlist 模式也可以科学上网了

closed 2020-05-04 by FishMagic

zhea55, 2020-05-04:
@FishMagic 
https://github.com/rufengsuixing/luci-app-adguardhome/issues/80

我今天研究了半天，有很多疑惑。

科学上网是肯定可以的。
国内去广告也没问题。

国外网站去广告，有问题。因为我们的adguard home部署在自己的内网里面，但是国外流量都是加密走的国外vps，vps在接受到流量请求以后，原封不动的把数据传输回我们本地，就是没有去广告的。

如果在passwall的chinaDNS-ng里面设置国外dns指向adguard，我之前一直这么用，没有多想，可以实现去广告的效果。

今天感觉有点不对劲，这么设置，虽然可以去广告，但是相当于远程主机，每个dns请求都向本地adguard发起，本地在把结果传输回去，效率好低，速度可想而知。

zhea55, 2020-05-04:
而且adguard home里面的平均dns处理时间，应该只算了接受到dns查询，向上级dns查询的时间。

从vps到本地，这个中间的时间，普通的线路至少200ms+