rufengsuixing/luci-app-adguardhome #83: 把最新luci-AdGuardHome源码编译进OpenWrt固件里，更新内核，无法启动。

jiatianxa, 2020-05-30:
固件版本 | OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f)

luci  AdGuardHome 源码编译OpenWrt固件。更新内核。无法启动，
但是直接用ipk在OpenWrt里安装就可以启动。
[image]

tujj99, 2020-06-05:
我也是这个版本OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f)，luci界面同样无法启动。
但是shell下可以手动启动。

sxx1314, 2020-06-05:
> 我也是这个版本OpenWrt R20.5.20 / LuCI Master（git-20.141.34051-f750b5f），luci界面也无法启动。
> 但是shell下可以手动启动。

master版（11）还是beta版（12）？

WROIATE, 2020-06-07:
> 固件版本 | OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f)
> 
> luci AdGuardHome 源码编译OpenWrt固件。更新内核。无法启动，
> 但是直接用ipk在OpenWrt里安装就可以启动。
> [image]

去/etc/init.d下面看看AdGuardHome的颜色，可能是权限问题。输入chmod +x /etc/init.d/AdGuardHome && service AdGuardHome restart试试

jiatianxa, 2020-06-07:
修改路由里"/etc/init.d/AdGuardHome"权限后就可以启动了

jiatianxa, 2020-06-07:
确实是权限问题，为何跟固件一起编译会造成权限不够的问题呢？

  

 
 
  
去/etc/init.d下面看看AdGuardHome的颜色，可能是权限问题。输入chmod +x /etc/init.d/AdGuardHome &amp;&amp; service AdGuardHome restart试试

WROIATE, 2020-06-09:
@jiatianxa 我估计的是编译的时候没有选择网络里面的adguardhome，然后编译的文件也没有设好权限
目前我的解决方法是，手动给adguardhome文件执行权限，然后把/usr/share/AdGuardHome下面的所有脚本给755权限

jiatianxa, 2020-06-09:
确实是没有用网络里的，我是下载后传到自己的库里的。看来是要修改权限后在传库里应该可以把？

tujj99, 2020-06-09:
> @jiatianxa 我估计的是编译的时候没有选择网络里面的adguardhome，然后编译的文件也没有设好权限
> 目前我的解决方法是，手动给adguardhome文件执行权限，然后把/usr/share/AdGuardHome下面的所有脚本给755权限

在luci-app-adguardhome的Makefile里面把二进制依赖加进去
DEPENDS:=+!wget&&!curl:wget +AdGuardHome

jiatianxa, 2020-06-09:
感谢指点

  

  
 DEPENDS:=+!wget&amp;&amp;!curl:wget +AdGuardHome

getkg, 2020-09-02:
> > @jiatianxa我估计的是编译的时候没有选择网络里面的adguardhome，然后编译的文件也没有设好权限
> > 目前我的解决方法是，手动给adguardhome文件执行权限，然后把/ usr / share / AdGuardHome下面的所有脚本给755权限
> 
> 在luci-app-adguardhome的Makefile里面把二进制依赖加
> 进去DEPENDS：= +！wget &&！curl：wget + AdGuardHome

请问你说的两种方式都要添加上去吗？

villaincoder, 2020-09-11:
参考了下其他app的Makefile，把Package/luci-app-adguardhome/install修改一下，试了一下没有问题
```
define Package/luci-app-adguardhome/install
    $(INSTALL_DIR) $(1)/usr/lib/lua/luci
	cp -pR ./luasrc/* $(1)/usr/lib/lua/luci
	$(INSTALL_DIR) $(1)/
	cp -pR ./root/* $(1)/
        # Fix: /etc/init.d/AdGuardHome: Permission denied
	$(INSTALL_BIN) ./root/etc/init.d/AdGuardHome $(1)/etc/init.d/AdGuardHome
	$(INSTALL_DIR) $(1)/usr/lib/lua/luci/i18n
	po2lmo ./po/zh-cn/AdGuardHome.po $(1)/usr/lib/lua/luci/i18n/AdGuardHome.zh-cn.lmo
endef
```

rufengsuixing, 2020-11-23:
你的源码大概率是解压来的吧的吧，可能丢了权限，建议还是git clone比较稳
关于这部分问题我尝试做下修正，毕竟快一年了，手头也没有调试环境了，可能就不打包了

yansinwang, 2021-11-17:
> > @jiatianxa 我估计的是编译的时候没有选择网络里面的adguardhome，然后编译的文件也没有设好权限
> > 目前我的解决方法是，手动给adguardhome文件执行权限，然后把/usr/share/AdGuardHome下面的所有脚本给755权限
> 
> 在luci-app-adguardhome的Makefile里面把二进制依赖加进去 DEPENDS:=+!wget&&!curl:wget +AdGuardHome

请问 把 DEPENDS:=+!wget&&!curl:wget +AdGuardHome 加到Makefile里的哪个位置？