rufengsuixing/luci-app-adguardhome #85: 从菜单进入adgurdhome报错

linxuankent, 2020-06-14:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/base'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua:49: attempt to concatenate local 'version' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

kongfl888, 2020-07-20:
+1
AdGuardHome/manual 出错，

完整日志如下
-------
Mon Jul 20 17:19:53 2020 daemon.err uhttpd[2529]: grep: /etc/AdGuardHome.yaml: No such file or directory
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tsection'.
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tvalue'.
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: A runtime error occurred: /usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:11: bad argument #1 to 'lines' (/tmp/resolv.conf.auto: No such file or directory)
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: stack traceback:
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[C]: in function 'lines'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:11: in function 'gen_template_config'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:46: in function 'cfgvalue'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[string "/usr/lib/lua/luci/view/cbi/tvalue.htm"]:1: in main chunk
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: stack traceback:
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[C]: in function 'error'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	(tail call): ?
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:1473: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:7: in main chunk
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: stack traceback:
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[C]: in function 'error'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/template.lua:97: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/dispatcher.lua:755: in function 'include'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[string "/usr/lib/lua/luci/view/cbi/tsection.htm"]:1: in main chunk
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: stack traceback:
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[C]: in function 'error'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	(tail call): ?
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: stack traceback:
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	[C]: in function 'error'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	(tail call): ?
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/dispatcher.lua:1450: in function '_cbi'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/dispatcher.lua:1023: in function 'dispatch'
Mon Jul 20 17:19:56 2020 daemon.err uhttpd[2529]: 	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>
-------

ludetro, 2020-08-03:
same issue

1179098689, 2020-08-15:
同