rufengsuixing/luci-app-adguardhome #87: feed下载软件源代码的时候出错

jerryliang122, 2020-06-29:
Updating feed 'adg' from 'https://github.com/rufengsuixing/luci-app-adguardhome' ...
Cloning into './feeds/adg'...
remote: Enumerating objects: 71, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 71 (delta 0), reused 44 (delta 0), pack-reused 0
Unpacking objects: 100% (71/71), done.
Create index file './feeds/adg.index' 
/home/jerry/lede/feeds/adg.tmp/info/.files-packageinfo.mk:1: *** target pattern contains no '%'.  Stop.
在下载源代码会出现
/home/jerry/lede/feeds/adg.tmp/info/.files-packageinfo.mk:1: *** target pattern contains no '%'.  Stop.的错误提示。
导致make menuconfig 看不到软件包

cokyrain, 2020-08-05:
我也遇到同样问题

openwrt@ubuntu:~/lede$ sed -i '$a src-git rufengsuixing https://github.com/rufengsuixing/luci-app-adguardhome' feeds.conf.default
openwrt@ubuntu:~/lede$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
已经是最新的。
Create index file './feeds/packages.index' 
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
已经是最新的。
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
已经是最新的。
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
已经是最新的。
Create index file './feeds/telephony.index' 
Updating feed 'freifunk' from 'https://github.com/freifunk/openwrt-packages.git' ...
已经是最新的。
Create index file './feeds/freifunk.index' 
Updating feed 'helloworld' from 'https://github.com/fw876/helloworld' ...
已经是最新的。
Create index file './feeds/helloworld.index' 
Updating feed 'rufengsuixing' from 'https://github.com/rufengsuixing/luci-app-adguardhome' ...
正克隆到 './feeds/rufengsuixing'...
remote: Enumerating objects: 71, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 71 (delta 0), reused 44 (delta 0), pack-reused 0
展开对象中: 100% (71/71), 101.35 KiB | 380.00 KiB/s, 完成.
Create index file './feeds/rufengsuixing.index' 
grep: feeds/rufengsuixing/Makefile:$(eval: 没有那个文件或目录
grep: $(call: 没有那个文件或目录
grep: BuildPackage,luci-app-adguardhome))/Makefile: 没有那个文件或目录
/home/openwrt/lede/feeds/rufengsuixing.tmp/info/.files-packageinfo.mk:1: *** 目标模式不含有“%”。 停止。
Collecting target info: done

390948623, 2020-10-09:
同上

AdoShan, 2022-02-19:
我的是代码都下载完成。
但是就是提示Makefiles文件夹不存在。
用的是Action
文件里面是这样写的
`echo 'src-git adguardhome https://github.com/rufengsuixing/luci-app-adguardhome.git' >>feeds.conf.default`
[文件如下](https://raw.githubusercontent.com/AdoShan/Build_OpenWrt/main/diy-part1.sh)
跑的时候就这样了
`Collecting target info: doneCreate index file './feeds/adguardhome.index' 
grep: feeds/adguardhome/Makefile:$(eval: No such file or directory
grep: $(call: No such file or directory
grep: BuildPackage,luci-app-adguardhome))/Makefile: No such file or directory
/workdir/openwrt/feeds/adguardhome.tmp/info/.files-packageinfo.mk:1: *** target pattern contains no '%'.  Stop.`
连上机器之后也是进不去这个文件夹。
`-bash: cd: Makefile: Not a directory`

ZqinKing, 2022-03-21:
相同问题，无法编译

ZqinKing, 2022-03-21:
变通了一下，用src-link解决了。

AdoShan, 2022-03-21:
> 变通了一下，用src-link解决了。
在哪改的哇？

AdoShan, 2022-03-21:
云编译无解了

先本地创建个问渐渐，git clone项目道本地
feeds.conf.default中添加 src-link adguardhome 本地路径/luci-app-adguardhome

siasjack, 2023-04-19:
> 变通了一下，用src-link解决了。

src-link datacenter git@192.168.0.99:kingkong/datacenter.git

我是这么添加的，实际上并未下载git库的代码
链接也是无效的

楼主说的方法仅适用于代码已经存在的情况下