rufengsuixing/luci-app-adguardhome #90: 如何才能得到 AdGuardHome.yaml ？

yansinwang, 2020-07-23:
### 提交之前

* **AdGuard Home luci的版本:**
 1.8-11
* **AdGuard Home server的版本:**
  指的是这个吗？
/usr/bin/wget-ssl

Local version: v0.102.0, cloud version: v0.102.0.
You're already using the latest version.

* **操作系统和版本:**

主机型号 | HuaWei HG255D (CpuMark : 836.199852 Scores)
架构 | Ralink RT3052 id:1 rev:3 x 1
固件版本 | OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)
内核版本 | 5.4.51

### 希望的执行结果
希望能找到配置文件，AdGuardHome.yaml

### 实际的执行结果

找不到AdGuardHome.yaml文件
awk: /etc/AdGuardHome.yaml: No such file or directory

首次安装
/usr/bin/wget-ssl

Failed to check latest version, please try again later.
/usr/bin/wget-ssl
Local version: ., cloud version: v0.102.0.
Updating core...
start download
--2020-07-22 19:42:02--  https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.102.0/AdGuardHome_linux_mipsle.tar.gz
Resolving github.com... 52.74.223.119
Connecting to github.com|52.74.223.119|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/34b81800-96d2-11ea-9034-50db5429fd7c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200722%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200722T114020Z&X-Amz-Expires=300&X-Amz-Signature=1906cc3e6fd9468d1fb73980b949092eea82550577a1dc074baf781dad48f49f&X-Amz-SignedHeaders=host&actor_id=0&repo_id=62712899&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_mipsle.tar.gz&response-content-type=application%2Foctet-stream [following]
--2020-07-22 19:42:06--  https://github-production-release-asset-2e65be.s3.amazonaws.com/62712899/34b81800-96d2-11ea-9034-50db5429fd7c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200722%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200722T114020Z&X-Amz-Expires=300&X-Amz-Signature=1906cc3e6fd9468d1fb73980b949092eea82550577a1dc074baf781dad48f49f&X-Amz-SignedHeaders=host&actor_id=0&repo_id=62712899&response-content-disposition=attachment%3B%20filename%3DAdGuardHome_linux_mipsle.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com... 52.216.179.235
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|52.216.179.235|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6366599 (6.1M) [application/octet-stream]
Saving to: '/tmp/AdGuardHomeupdate/AdGuardHome_linux_mipsle.tar.gz'

     0K .......... .......... .......... .......... ..........  0% 90.3K 68s
    50K .......... .......... .......... .......... ..........  1%  582K 39s
手动省略
  6150K .......... .......... .......... .......... .......... 99% 65.0K 0s
  6200K .......... .......                                    100% 85.7K=31s

2020-07-22 19:42:38 (202 KB/s) - '/tmp/AdGuardHomeupdate/AdGuardHome_linux_mipsle.tar.gz' saved [6366599/6366599]

download success start copy
start copy
awk: /etc/AdGuardHome.yaml: No such file or directory
AdGuardHome service disabled
Command failed: Not found
rm: can't remove '/usr/bin/AdGuardHome/AdGuardHome': No such file or directory

之后 手动点击 更新核心
[image]

更新按钮下的提示变为：核心版本:v0.102.0 没有配置文件no core没有核心
     root@OpenWrt:/usr/share/AdGuardHome# sh  update_core.sh
/usr/bin/wget-ssl

Local version: v0.102.0, cloud version: v0.102.0.
You're already using the latest version.
[image]

[image]

同时c存在的另外一个小问题：
[image]

<h2 name="content">500 内部服务器错误</h2>
<p>对不起，服务器遇到未知错误。</p>
<pre class="error500">/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/manual'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tvalue'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:11: bad argument #1 to 'lines' (/tmp/resolv.conf.auto: No such file or directory)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120></pre>

   <footer>
    <a href="https://github.com/openwrt/luci">Powered by LuCI Master (git-20.191.36863-eee6bae)</a> / OpenWrt  R20.7.20
    
   </footer>
   </div>
  </div>
 </body>
</html>

### 日志(重要)
[AdGuardHome_update.log](https://github.com/rufengsuixing/luci-app-adguardhome/files/4963681/AdGuardHome_update.log)

### 截图

### 更多的信息

sdrpsps, 2020-07-23:
跟你一样

InoryLwc, 2020-07-27:
问题加一，我不知道影响不影响启动，我搞了半天都用不了

InoryLwc, 2020-07-27:
我在里面找到了yaml文件，终于能用了
https://github.com/rufengsuixing/luci-app-adguardhome/blob/master/root/usr/share/AdGuardHome/AdGuardHome_template.yaml

sdrpsps, 2020-07-27:
> 我在里面找到了yaml文件，终于能用了
> https://github.com/rufengsuixing/luci-app-adguardhome/blob/master/root/usr/share/AdGuardHome/AdGuardHome_template.yaml

谢谢！

yanglibin, 2021-01-24:
> 我在里面找到了yaml文件，终于能用了
> https://github.com/rufengsuixing/luci-app-adguardhome/blob/master/root/usr/share/AdGuardHome/AdGuardHome_template.yaml

经过实测, 此方法不可用.

yanglibin, 2021-01-24:
1.  去 AdGuardHome 的github [官方项目](https://github.com/AdguardTeam/AdGuardHome/) 下载最新的 AdGuardHome.  ( 注意: 树莓派 openwrt 中的 AdGuardHome 为 armv7 版本 <AdGuardHome_linux_armv7.tar.gz> )
2.  将 AdGuardHome_linux_armv7.tar.gz 解压, 得到  AdGuardHome 
3.  将 AdGuardHome 通过 openwrt 的 "文件传输"  ("系统" -> "文件传输" ) 上传到 openwrt 中
4.  登录到 openwrt 上进入如下操作:
    a.  `chmod +x  /tmp/upload/AdGuardHome ` 
    b.  `cp /tmp/upload/AdGuardHome   /usr/bin/AdGuardHome/AdGuardHome ` 
    c.  `cp /usr/share/AdGuardHome/AdGuardHome_template.yaml /etc/AdGuardHome.yaml` 

5. 在 openwrt 界面启用 AdGuard Home, 并刷新页面
[image]
[image]

yanglibin, 2021-01-24:
1.  去 AdGuardHome 的github [官方项目](https://github.com/AdguardTeam/AdGuardHome/) 下载最新的 AdGuardHome.  ( 注意: 树莓派 openwrt 中的 AdGuardHome 为 armv7 版本 <AdGuardHome_linux_armv7.tar.gz> )
2.  将 AdGuardHome_linux_armv7.tar.gz 解压, 得到  AdGuardHome 
3.  将 AdGuardHome 通过 openwrt 的 "文件传输"  ("系统" -> "文件传输" ) 上传到 openwrt 中
4.  登录到 openwrt 上进入如下操作:
    a.  `chmod +x  /tmp/upload/AdGuardHome ` 
    b.  `cp /tmp/upload/AdGuardHome   /usr/bin/AdGuardHome/AdGuardHome ` 
    c.  `cp /usr/share/AdGuardHome/AdGuardHome_template.yaml /etc/AdGuardHome.yaml` 

5. 在 openwrt 界面启用 AdGuard Home, 并刷新页面

[ 以上方法, 亲测试有效 ]

qiuwl88, 2021-02-22:
> 1. 去 AdGuardHome 的github [官方项目](https://github.com/AdguardTeam/AdGuardHome/) 下载最新的 AdGuardHome.  ( 注意: 树莓派 openwrt 中的 AdGuardHome 为 armv7 版本 <AdGuardHome_linux_armv7.tar.gz> )
> 2. 将 [...]

用此方法后不提示核心版本:v0.102.0 没有配置文件no core没有核心，但启用不了，显示未运行未重定向

Tuuure, 2021-03-14:
遇到了同样的问题，以下方法可解决
cp /usr/share/AdGuardHome/AdGuardHome_template.yaml /etc/AdGuardHome.yaml

daydayup763, 2021-12-10:
我也是这个问题。你们都解决了嘛

masx200, 2024-03-28:
升级到核心版本:v0.107.46,之前的配置文件又不能用了!
```yaml
http:
  pprof:
    port: 3000
    enabled: false
  address: 0.0.0.0:3000
  session_ttl: 720h
users:
  - name: admin
    password: $2y$10$0iNmfiUYHLScukNxVvxm2ehVkJVj4SVyPGXgqjCw7/abK/FvmMWZG
auth_attempts: 5
block_auth_min: 15
http_proxy: ""
language: ""
theme: auto
dns:
  bind_hosts:
    - 0.0.0.0
  port: 5553
  anonymize_client_ip: false
  ratelimit: 20
  ratelimit_subnet_len_ipv4: 24
  ratelimit_subnet_len_ipv6: 56
  ratelimit_whitelist: []
  refuse_any: true
  upstream_dns:
    - quic://8fc1f6b9.d.adguard-dns.com
    - https://dns10.quad9.net/dns-query
    - https://deno-dns-over-https-server.deno.dev/dns-query
    - quic://unfiltered.adguard-dns.com
    - https://doh.cleanbrowsing.org/doh/security-filter/
    - https://doh.360.cn/dns-query
    - https://d.adguard-dns.com/dns-query/8fc1f6b9
    - tls://8fc1f6b9.d.adguard-dns.com
    - 2a10:50c0:c000::8fc1:f6b9
    - 94.140.14.59
  upstream_dns_file: ""
  bootstrap_dns:
    - 9.9.9.10
    - 149.112.112.10
    - 2620:fe::10
    - 2620:fe::fe:10
  fallback_dns: []
  upstream_mode: parallel
  fastest_timeout: 1s
  allowed_clients: []
  disallowed_clients: []
  blocked_hosts:
    - version.bind
    - id.server
    - hostname.bind
  trusted_proxies:
    - 127.0.0.0/8
    - ::1/128
  cache_size: 4194304
  cache_ttl_min: 600
  cache_ttl_max: 0
  cache_optimistic: true
  bogus_nxdomain: []
  aaaa_disabled: false
  enable_dnssec: false
  edns_client_subnet:
    custom_ip: ""
    enabled: false
    use_custom: false
  max_goroutines: 300
  handle_ddr: true
  ipset: []
  ipset_file: ""
  bootstrap_prefer_ipv6: false
  upstream_timeout: 10s
  private_networks: []
  use_private_ptr_resolvers: true
  local_ptr_upstreams: []
  use_dns64: false
  dns64_prefixes: []
  serve_http3: false
  use_http3_upstreams: false
  serve_plain_dns: true
  hostsfile_enabled: true
tls:
  enabled: false
  server_name: desktop-5rh4po7.manx-sun.ts.net
  force_https: false
  port_https: 3443
  port_dns_over_tls: 853
  port_dns_over_quic: 853
  port_dnscrypt: 0
  dnscrypt_config_file: ""
  allow_unencrypted_doh: false
  certificate_chain: ""
  private_key: ""
  certificate_path: C:\\Users\\Administrator\\tailscale-certs\\cert.pem
  private_key_path: C:/Users/Administrator/tailscale-certs/key.pem
  strict_sni_check: false
querylog:
  dir_path: ""
  ignored: []
  interval: 2160h
  size_memory: 1000
  enabled: true
  file_enabled: true
statistics:
  dir_path: ""
  ignored: []
  interval: 24h
  enabled: true
filters:
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_1.txt
    name: AdGuard DNS filter
    id: 1
  - enabled: false
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_2.txt
    name: AdAway Default Blocklist
    id: 2
whitelist_filters: []
user_rules: []
dhcp:
  enabled: false
  interface_name: ""
  local_domain_name: lan
  dhcpv4:
    gateway_ip: ""
    subnet_mask: ""
    range_start: ""
    range_end: ""
    lease_duration: 86400
    icmp_timeout_msec: 1000
    options: []
  dhcpv6:
    range_start: ""
    lease_duration: 86400
    ra_slaac_only: false
    ra_allow_slaac: false
filtering:
  blocking_ipv4: ""
  blocking_ipv6: ""
  blocked_services:
    schedule:
      time_zone: UTC
    ids: []
  protection_disabled_until: null
  safe_search:
    enabled: false
    bing: true
    duckduckgo: true
    google: true
    pixabay: true
    yandex: true
    youtube: true
  blocking_mode: default
  parental_block_host: family-block.dns.adguard.com
  safebrowsing_block_host: standard-block.dns.adguard.com
  rewrites: []
  safebrowsing_cache_size: 1048576
  safesearch_cache_size: 1048576
  parental_cache_size: 1048576
  cache_time: 30
  filters_update_interval: 24
  blocked_response_ttl: 10
  filtering_enabled: true
  parental_enabled: false
  safebrowsing_enabled: false
  protection_enabled: true
clients:
  runtime_sources:
    whois: true
    arp: true
    rdns: true
    dhcp: true
    hosts: true
  persistent: []
log:
  file: ""
  max_backups: 0
  max_size: 100
  max_age: 3
  compress: false
  local_time: false
  verbose: false
os:
  group: ""
  user: ""
  rlimit_nofile: 0
schema_version: 28

```