rufengsuixing/luci-app-adguardhome #91: 核心版本更新到最新版本0.103.3 报错:

LiangJunheng, 2020-07-25:
核心版本更新到最新版本0.103.3 报错:  

/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/base'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua:49: attempt to concatenate local 'version' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

h31105, 2020-07-25:
openwrt adguard 的更新界面无法更新，但通过adguard面板更新到最新版
但更新后，openwrt adguard 页面无法识别新版本。

LiangJunheng, 2020-07-25:
> openwrt adguard 的更新界面无法更新，可以通过adguard面板更新到最新版
> 但更新后，openwrt adguard 页面无法识别新版本。（不影响使用

这个我是知道的，就是想反馈一下

LiangJunheng, 2020-07-25:
> openwrt adguard 的更新界面无法更新，可以通过adguard面板更新到最新版
> 但更新后，openwrt adguard 页面无法识别新版本。（不影响使用

希望作者可以修复下

ameshkov, 2020-07-25:
@rufengsuixing hi! we changed the AGH version string, you need to strip the "v" prefix when comparing versions

cnstark, 2020-07-31:
应该是因为没有version变量导致的，我把/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua 49行注释掉并重新对e赋值，lua界面就可以打开了
[image]
不过影响了版本号的获取，但插件可以正常使用
[image]

LiangJunheng, 2020-08-03:
> 应该是因为没有version变量导致的，我把/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua 49行注释掉并重新对e赋值，lua界面就可以打开了
> [image]
> 不过影响了版本号的获取，但插件可以正常使用
> [image]

> 应该是因为没有version变量导致的，我把/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua 49行注释掉并重新对e赋值，lua界面就可以打开了
> [image]
> 不过影响了版本号的获取，但插件可以正常使用
> [image]

可以，完美解决进入界面问题

closed 2020-08-03 by LiangJunheng