rufengsuixing/luci-app-adguardhome #95: 更新版本后无法显示核心版本号

james-code-dev, 2020-08-07:
用Luci更新核心版本后，在管理介面里不显示核心版本，但是程序运行不受影响。如下图：
[image]

我用的是L大的OpenWrt R20.4.8.

dl250, 2020-08-07:
我的也是

xiaogtiang, 2020-08-07:
我的是ad可以更新，smartdns无法启动

james-code-dev, 2020-08-09:
> 用Luci更新核心版本后，在管理介面里不显示核心版本，但是程序运行不受影响。如下图：
> [image]
> 
> 我用的是L大的OpenWrt R20.4.8.

今天找到解决办法了，把Luci里自带更新源注释掉，然后手动添加以下地址：
https://static.adguard.com/adguardhome/edge/AdGuardHome_linux_${Arch}.tar.gz
[image]

原因是楼主这个Luci识别不了Beta版本的核心，所以显示异常了，经过我们手动替换稳定源即可解决。

ziyourenchina, 2020-08-24:
替换了 还是不可以哎

james-code-dev, 2020-08-25:
谢谢Kongfl888，已经可以正常显示版本号了。