rufengsuixing/luci-app-adguardhome #96: 最新版AdGuardHome无法获取版本号，导致应用无法安装使用

chingxuds, 2020-08-09:
全新安装后更新核心，更新完之后无法获取核心版本号，导致页面访问报错：
```
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute function dispatcher target for entry '/admin/services/AdGuardHome'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/AdGuardHome/base'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua:49: attempt to concatenate local 'version' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
```

经检查，新版AdGuardHome核心版本信息是`version 0.103.3`，但是获取版本信息的命令是`grep -m 1 -E 'v[0-9.]+' -o`，因此已无法正常获取版本信息，需要更新版本获取命令。

chenzch, 2021-01-08:
usr/share/AdGuardHome/update_core.sh Line 30 : 
now_ver="$($binpath -c /dev/null --check-config 2>&1| grep -m 1 -E 'v[0-9.]+' -o)" 
change to
now_ver="$($binpath -c /dev/null --check-config 2>&1| grep -m 1 -E 'version [0-9.]+' -o|sed 's/version /v/g')"

chenzch, 2021-01-20:
/usr/lib/lua/luci/model/cbi/AdGuardHome/base.lua Line 40:
local tmp=luci.sys.exec(binpath.." -c /dev/null --check-config 2>&1| grep -m 1 -E 'version [0-9.]+' -o| sed 's/version /v/g'")