rufengsuixing/luci-app-adguardhome #105: AGH配合SSR+，白名单失效。

kkfuzi, 2020-09-11:
- [*] 我用着最新版的 luci-app-adguardhome
- [*] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  * <1.8-11->
* **操作系统和版本:**
  * <OpenWrt R20.8.27>

软路由，SSR+搭配ADH，SSR+模式 绕过大陆IP，DNS解析是PDNSD
发现如果ADH设置成“使用53端口替换 dnsmasq” 模式， SSR+的访问控制/白名单 无效，设置成白名单的国外域名依然走代理。
如果ADH设置成“作为dnsmasq的上游服务器” 就好了。。。

但用上游服务器模式就没法统计局域网客户端信息和自定义客户端的广告过滤

不知道有什么解决方法，谢谢！

kkfuzi, 2020-09-13:
关于我的测试方案，三个测IP的网站：
http://test-ipv6.com/
https://whatismyipaddress.com/
http://www.882667.com/

分别把域名随机加入白名单/黑名单，测试下来只有“dns重定向-作为dnsmasq的上游服务器”模式工作正常（连接符合白名单黑名单设置）。其他两种重定向模式，不管三个网站放在白名单还是黑名单，全部走代理。

mysteryman2045, 2020-09-18:
跟luci-app-adguardhome无关，需要核心支持。官方的v0.104.0-beta1开发进程里边：“7931e50 + DNS: add "ipset" configuration setting”就是在做这个优化，只能坐等了。

kkfuzi, 2020-09-19:
> 跟luci-app-adguardhome无关，需要核心支持。官方的v0.104.0-beta1开发进程里边：“7931e50 + DNS: add "ipset" configuration setting”就是在做这个优化，只能坐等了。

谢谢回复！