rufengsuixing/luci-app-adguardhome #117: 停用或删除后，域名无法解析了（国内可以，外网出不去）

3xpert, 2021-02-20:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [√] 我用着最新版的 luci-app-adguardhome
- [√] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  *  luci-app-adguardhome_1.8-20200824
* **AdGuard Home server的版本:**
  *  v0.105.1
* **操作系统和版本:**
  * OpenWrt 19.07.3 x86 esxi

### 希望的执行结果
在停用或移除AGH后，系统能回复到原来的状态
### 实际的执行结果
esxi虚拟机上安装openwrt，通过shadowsocks-libev和chinadns来实现外网访问，安装luci-app-adguardhome后，设置dns重定向为”使用53端口替换dnsmasq“，AGH的上游服务器设置为本机127.0.0.1:5354，当停用AGH后（甚至移除），外网就无法访问了，显然是没有获得域名解释
### 日志(重要)

### 截图
shadowsocks-libev和chinadns的设置如下
[image]

<details><summary>截图:</summary>

恳请大佬解惑，谢谢！

</details>

### 更多的信息

bluebabes, 2022-03-11:
我也遇到了。。dns 不解析了。

qirenzhidao, 2023-04-22:
我也是，怎么回事，之前没问题。现在dns服务器好像没了，不解析了