rufengsuixing/luci-app-adguardhome #118: 请问openwrt中luci-app-adguardhome应该如何和SSRPlus配合使用呢？

dzhl, 2021-03-05:
对此一直很困惑，网上搜了些文章，但总感觉没有什么效果，请问大家有比较好的方案吗？

dzhl, 2021-03-10:
有人可以指导下吗？

dzhl, 2021-03-11:
等指导

kr513268, 2021-03-12:
关于ssr配合
方法一gfw代理：dns重定向-作为dnsmasq的上游服务器
方法二gfw代理：手动设置adh上游dns为自己即127.0.0.1:[自己监听的端口]，然后使用 dns重定向-使用53端口替换dnsmasq，（因为端口互换后就是dnsmasq为上游了）
方法三国外ip代理：任意重定向方式，adh加入gfw列表，开启计划任务定时更新gfw即可
方法四gfw代理：dns重定向-重定向53端口到AdGuardHome,设置adh上游dns 为127.0.0.1:53

dzhl, 2021-03-15:
> 关于ssr配合
> 方法一gfw代理：dns重定向-作为dnsmasq的上游服务器
> 方法二gfw代理：手动设置adh上游dns为自己即127.0.0.1:[自己监听的端口]，然后使用 dns重定向-使用53端口替换dnsmasq，（因为端口互换后就是dnsmasq为上游了）
> 方法三国外ip代理：任意重定向方式，adh加入gfw列表，开启计划任务定时更新gfw即可
> 方法四gfw代理：dns重定向-重定向53端口到AdGuardHome,设置adh上游dns 为127.0.0.1:53

谢谢耐心回答，我一般用gfw代理，请问这三种方式有优劣之分吗？

Kaze1027, 2021-06-17:
老老实实用方法一就好了，不用折腾了，还有这个问题可以关闭了，这不是issues