rufengsuixing/luci-app-adguardhome #122: 核心文件更新失败导致无法启动

hhl118, 2021-06-28:
遇到这个怎么解决？有人和我一样吗？

[image]<br /><a target='_blank' href='https://imgbb.com/'></a><br />

[image]

zwillhill, 2021-07-05:
缺少wget-ssl依赖包。

djdjfjfkeocj, 2024-07-01:
> 缺少wget-ssl依赖包。

在哪下依赖包呀

djdjfjfkeocj, 2024-07-05:
> 遇到这个怎么解决？有人和我一样吗？
> 
> [image]<br /><a target='_blank' href='https://imgbb.com/'></a><br />
> 
> [image]

重新再点一下就好了，应该是第一次没有那个文件夹然后它会自动创建，第二遍点更新就会成功