rufengsuixing/luci-app-adguardhome #126: 重定向选择作为dnsmasq上游，插件不起作用

yjlmiss, 2021-08-22:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [ ] 我用着最新版的 luci-app-adguardhome
- [ ] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  1.8-11
* **AdGuard Home server的版本:**
 v0.106.3
* **操作系统和版本:**
 OpenWrt R21.8.6

### 希望的执行结果
配置接口dns为127.0.0.1，dhcp\dns为127.0.0.1:adguard端口，设置重定向方式为选择作为dnsmasq上游服务器，adguard上游服务器设置223.5.5.5，启用成功，希望正常上网解析dns，拦截配置网址。
### 实际的执行结果
配置作为dnsmasq的上游，上不了网。无法解析dns
### 日志(重要)

cat AdGuardHome_update.log 

Local version: v0.106.3, cloud version: v0.106.3.
You're already using the latest version.

### 截图

/etc/init.d/AdGuardHome start
workdir is a ubifs filesystem
AdGuardHome service enabled
luci enable switch=1

<details><summary>截图:</summary>

</details>

### 更多的信息

yjlmiss, 2021-08-22:
不知道是哪里配置错了？求指教。

LeviMarvin, 2022-03-20:
我也是！

lemchen, 2022-04-02:
请正确配置 adguard home 的 dns 端口，然后在dnsmasq 配置里 dns 转发到 adguard home的 端口
举个例子：
adguard home dns 端口设置为5353，那么在dnsmasq 里需要配置dns转发 到 127.0.0.1:5353，dnsmasq本身的端口还是53