rufengsuixing/luci-app-adguardhome #130: awk: /etc/AdGuardHome.yaml: No such file or directory

zhangguanzhang, 2022-01-09:
```
root@OpenWrt:~# opkg list | grep -i adg
kmod-usb-gadget - 5.4.162-1 - Kernel support for USB Gadget mode
luci-app-adguardhome - git-22.007.21624-d880280-1 - LuCI support for AdGuardHome
luci-i18n-adguardhome-zh-cn - git-22.007.21624-d880280-1 - Translation for luci-app-adguardhome - 简体中文 (Simplified Chinese)
root@OpenWrt:~# opkg install luci-app-adguardhome
Installing luci-app-adguardhome (git-22.007.21624-d880280-1) to root...
Downloading file:///local_feed/luci-app-adguardhome_git-22.007.21624-d880280-1_all.ipk
Configuring luci-app-adguardhome.
awk: /etc/AdGuardHome.yaml: No such file or directory
```

zhangguanzhang, 2022-01-09:
hack下后编译的ipk文件就能解决
```
    if [ -d feeds/others/luci-app-adguardhome ];then
        sed -i '/configpath/s#/etc/AdGuardHome.yaml#/etc/config/AdGuardHome.yaml#' feeds/others/luci-app-adguardhome/root/etc/config/AdGuardHome
    fi
    # https://github.com/rufengsuixing/luci-app-adguardhome/issues/130
    SED_NUM=$(awk '/^start_service/,/configpath/{a=NR}END{print a}' feeds/others/luci-app-adguardhome/root/etc/init.d/AdGuardHome)
    sed -i "$SED_NUM"'a [ ! -f "${configpath}" ] && cp /usr/share/AdGuardHome/AdGuardHome_template.yaml ${configpath}' feeds/others/luci-app-adguardhome/root/etc/init.d/AdGuardHome
```

zhangguanzhang, 2022-01-10:
卸载也会报错 `Command failed: Not found`
```
root@OpenWrt:~# opkg remove luci-app-adguardhome
Removing package luci-app-adguardhome from root...
AdGuardHome service disabled
AdGuardHome service disabled

Not deleting modified conffile /etc/config/AdGuardHome.
```

karnadii, 2022-04-03:
do you find solution? I also have the same problem

zhangguanzhang, 2022-04-04:
> sed -i '/configpath/s#/etc/AdGuardHome.yaml#/etc/config/AdGuardHome.yaml#' feeds/others/luci-app-adguardhome/root/etc/config/AdGuardHome

try this command

```
SED_NUM=$(awk '/^start_service/,/configpath/{a=NR}END{print a}' /etc/init.d/AdGuardHome)
sed -i "$SED_NUM"'a [ ! -f "${configpath}" ] && cp /usr/share/AdGuardHome/AdGuardHome_template.yaml ${configpath}' feeds/others/luci-app-adguardhome/root/etc/init.d/AdGuardHome
```