rufengsuixing/luci-app-adguardhome #132: 使用53端口替换dnsmasq时设备无法得到 ipv4

labarry, 2022-02-23:


labarry, 2022-02-25:
问题已经解决：
[image]
[image]

Travel2Here, 2022-04-20:
哥们，dnsmasq是干嘛的你都没搞懂？（里面有dns解析管理和DHCP功能）你都把它抛弃了 它当然不分配ip，你现在要做的是在ADG里面的DHCP里开启分配。

guagua0, 2022-08-22:
> 哥们，dnsmasq是干嘛的你都没搞懂？（里面有dns解析管理和DHCP功能）你都把它抛弃了 它当然不分配ip，你现在要做的是在ADG里面开启它的DHCP分配。

我就设置了一次 再也弄不会来了,现在dnsmasq里端口要设置成52 并开启dns转发才能用 不然所有ip都解析不了,
也是没有ipv4dns服务器 都把插件卸载了也不行

a757814283, 2022-11-13:
@guagua0 兄弟，我也是这个问题困扰了很久，一直以为是防火墙还是什么的问题，但感谢你提供了思路。

主要的问题确实是DHCP分配的时候不分配DNS服务器，这个是在Openwrt19的时候是没有的，不过后来好像改了，就是DHCP分配ip的时候可能是跟上端口了，然后大部分设备是不支持自定义dns端口的，也就不接受分配的dns。
或者是Dnsmasq发现改了DNS服务器端口了就干脆不分配DNS服务了。/个人的一点yy，欢迎指正

我的解决办法就是在dhcp服务器指定dns服务器，找到
LAN→DHCP服务器→高级设置→DHCP 选项→填入 6,网关地址
比如：6,192.168.1.1

ps：楼上那老哥好像确实把自己当那么回事了。

hbrtv, 2022-11-25:
> @guagua0 兄弟，我也是这个问题困扰了很久，一直以为是防火墙还是什么的问题，但感谢你提供了思路。
>
> 主要的问题确实是DHCP分配的时候不分配DNS服务器，这个是在Openwrt19的时候是没有的，不过后来好像改了，就是DHCP分配ip的时候可能是跟上端口了，然后大部分设备是不支持自定义dns端口的，也就不接受分配的dns。 [...]

查了一晚上，这个是正解。