rufengsuixing/luci-app-adguardhome #133: 无法正常安装和使用

LeviMarvin, 2022-03-19:
### 问题详细信息

* **AdGuard Home luci的版本:**
  * 1.8-11
* **AdGuard Home server的版本:**
  * v107.5
* **操作系统和版本:**
  * OpenWrt 21.02.2

### 希望的执行结果
1. 正确显示快速配置文件
2. 正确更新核心
### 实际的执行结果
1. `
<script type="text/javascript" src="/luci-static/resources/promis.min.js?v=git-22.052.50801-31a27f3"></script>
<script type="text/javascript" src="/luci-static/resources/luci.js?v=git-22.052.50801-31a27f3"></script>
<script type="text/javascript">
	L = new LuCI({"apply_rollback":90,"resource":"\/luci-static\/resources","media":"\/luci-static\/bootstrap","documentroot":"\/www","pathinfo":"\/admin\/services\/AdGuardHome\/manual","apply_display":1.5,"requestpath":["admin","services","AdGuardHome","manual"],"ubuspath":"\/ubus\/","scriptname":"\/cgi-bin\/luci","dispatchpath":["admin","services","AdGuardHome","manual"],"sessionid":"1beeb5a64801617f487c7bda88c9c91c","token":"018946e13fe9fd4f0dd9393fc132b140","apply_timeout":5,"apply_holdoff":4,"nodespec":{"wildcard":true,"satisfied":true,"title":"Manual Config","order":3,"action":{"post":{"cbi.submit":true},"type":"cbi","path":"AdGuardHome\/manual"}},"pollinterval":5});
</script>

<h2 name="content">500 内部服务器错误</h2>
<p>对不起，服务器遇到未知错误。</p>
<pre class="error500">/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tsection'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tvalue'.
A runtime error occurred: /usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:11: bad argument #1 to 'lines' (/tmp/resolv.conf.auto: No such file or directory)
stack traceback:
	[C]: in function 'lines'
	/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:11: in function 'gen_template_config'
	/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:46: in function 'cfgvalue'
	[string "/usr/lib/lua/luci/view/cbi/tvalue.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:1473: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:7: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:754: in function 'include'
	[string "/usr/lib/lua/luci/view/cbi/tsection.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478></pre>

		
		</div>
		<footer>
			<span>
				<a href="https://github.com/openwrt/luci">Powered by LuCI openwrt-21.02 branch (git-22.052.50801-31a27f3)</a> / OpenWrt 21.02.2 r16495-bf0c965af0
			</span>
			<ul class="breadcrumb pull-right" id="modemenu" style="display:none"></ul>
		</footer>
		<script type="text/javascript">L.require('menu-bootstrap')</script>
		
	</body>
</html>

`
2. ` /usr/bin/curl
Local version: ., cloud version: v0.107.5.
Updating core...
start download
curl: (7) Error
download failed try another download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (35) SSL_connect failed with error -308: error state on socket
download failed try another download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
  0 6217k    0 28467    0     0   8926      0  0:11:53  0:00:03  0:11:50  9932
  1 6217k    1 81891    0     0  20699      0  0:05:07  0:00:03  0:05:04 22540
  4 6217k    4  287k    0     0  58702      0  0:01:48  0:00:05  0:01:43 62740
 20 6217k   20 1263k    0     0   209k      0  0:00:29  0:00:06  0:00:23  280k
 55 6217k   55 3455k    0     0   484k      0  0:00:12  0:00:07  0:00:05  681k
 81 6217k   81 5039k    0     0   621k      0  0:00:10  0:00:08  0:00:02 1019k
 96 6217k   96 6015k    0     0   656k      0  0:00:09  0:00:09 --:--:-- 1138k
100 6217k  100 6217k    0     0   657k      0  0:00:09  0:00:09 --:--:-- 1335k
download success start copy
start copy
awk: /etc/AdGuardHome.yaml: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
AdGuardHome service disabled
Command failed: Not found
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
grep: /etc/crontabs/root: No such file or directory
awk: /etc/AdGuardHome.yaml: No such file or directory
Succeeded in updating core.
Local version: v0.107.5, cloud version: v0.107.5.`
虽然最后显示成功，但实际没有成功，刷新页面依然为“no core”
### 日志(重要)

### 截图

<details><summary>截图:</summary>
[image]

</details>

### 更多的信息
1.8-9版本也是这样

lemchen, 2022-04-02:
安装 luci-compat 了吗？

LeviMarvin, 2022-04-02:
安装了

WilsonXu0512, 2022-04-09:
好像是最新版本的AdguardHome把执行路径和配置路径都改了，这边的Luci包还没改，所以读不到配置（现在改为/opt/AdguardHome/AdguardHome.yaml，原来是/etc/AdguardHome.yaml）
目前我的解决方案是保存好旧版的yaml配置文件，重新手动安装AdguardHome,再用Winscp进去新路径覆盖掉原始配置文件。操作界面用XXX.XXX.XXX.XXX:3000的方式进，不用图形界面了

LeviMarvin, 2022-04-12:
我不再用 LuCI 进行操作，按照官方教程安装、替换 DNS 解析后，直接访问 AdGuradHome 的 IP:端口 的 WEB 控制台进行操作。
这是目前最好的解决方案。

gsxzn163, 2022-05-22:
[image]
我的也这样

lsq726, 2023-03-27:
> 

> 我不再用 LuCI 进行操作，按照官方教程安装、替换 DNS 解析后，直接访问 AdGuradHome 的 IP:端口 的 WEB 控制台进行操作。 这是目前最好的解决方案。

官方  openwrt  下怎么操作？

lsq726, 2023-03-28:
> 我不再用 LuCI 进行操作，按照官方教程安装、替换 DNS 解析后，直接访问 AdGuradHome 的 IP:端口 的 WEB 控制台进行操作。 这是目前最好的解决方案。

请教官方教程安装是怎么操作？