rufengsuixing/luci-app-adguardhome #134: 设备重启后ADH不能正常启动和重定向

ssrubsd, 2022-04-13:
### 提交之前

在你提交问题之前请回答以下问题
**你可以删掉提交之前这个部分**

- [√] 我用着最新版的 luci-app-adguardhome
- [√] 我检查了没有人提交同样的问题

### 问题详细信息

* **AdGuard Home luci的版本:**
  核心版本:v0.108.0
* **AdGuard Home server的版本:**
 核心版本:v0.108.0
* **操作系统和版本:**
OpenWrt R22.4.1

### 希望的执行结果
在设备重启后ADH能正常运行
### 实际的执行结果
设备重启后刚开始时ADH能正常运行和重定向，随后自动关闭，手动启动后ADH的配置一切正常
### 日志(重要)
系统日志：user.notice root: AdGuardHome no process in 10s cancel redirect
ADH日志：[info] Received signal "hangup"
### 截图

<details><summary>截图:</summary>

</details>

### 更多的信息
此问题好像只在ADH设置为替换dnsmasq模式时才会出现

wy580477, 2022-07-01:
我也遇到了，凑合解决办法是禁止AdGuardHome服务自启，rc.local中加入“service AdGuardHome start”命令。

不过我现在不用luci-app-adguardhome了，启动脚本搞的很复杂改一大堆设置，其实我只需要启动adguardhome程序。

xain, 2022-08-24:
我也遇到了同样的问题。

dahounet, 2022-10-05:
看了一下 https://github.com/rufengsuixing/luci-app-adguardhome/blob/master/root/etc/init.d/AdGuardHome

似乎是启动脚本里面写死了“/usr/bin/AdGuardHome/AdGuardHome”这个路径用来在开机启动的时候判断程序是否启动成功，但是现在的路径默认是“/usr/bin/AdGuardHome”，导致误判断把AdGuardHome配置给重置并关掉了。

解决：
1.到/usr/bin/里删掉“AdGuardHome”文件（若有这个文件的话）
2.再在/usr/bin/这里建个“AdGuardHome”文件夹
3.到AdGuard Home 的设置页面把执行文件路径由“/usr/bin/AdGuardHome”改为“/usr/bin/AdGuardHome/AdGuardHome”
4.下载核心
5.启用AdGuard Home后再重启OpenWrt系统试试，应该可以正常启动了。

lqh280, 2022-11-14:
binpath '/usr/bin/AdGuardHome/AdGuardHome'
workdir '/usr/bin/AdGuardHome'
我看了其它packages版本，包括别人编译后的，比较后：
binpath  '/usr/bin/AdGuardHome'，这个 /usr/bin目录，没有AdGuardHome目录的。
workdir '/etc/AdGuardHome'，workdir 都没在/usr/bin，是在/etc目录的。
如果是自己编译，建议将luci-app-adguardhome目录，所有文件，都检查一下，有不少文件需要修改的。

Amanoshoyichi, 2022-12-03:
是的我也遇到了这个情况。但是小白不太会用命令行。

audyfan, 2022-12-16:
遇到了类似的问题。附上日志

2022/12/16 16:13:23.203843 [fatal] initing hosts container: hosts container: path at index 0: stat etc/hosts: os: DirFS with empty root
2022/12/16 16:13:23.203516 [info] tls: using default ciphers
2022/12/16 16:13:23.197863 [info] AdGuard Home, version v0.107.18
2022/12/16 16:13:18.165445 [fatal] initing hosts container: hosts container: path at index 0: stat etc/hosts: os: DirFS with empty root
2022/12/16 16:13:18.164989 [info] tls: using default ciphers
2022/12/16 16:13:18.155966 [info] AdGuard Home, version v0.107.182022/12/16 16:13:43.570213 [fatal] initing hosts container: hosts container: path at index 0: stat etc/hosts: os: DirFS with empty root
2022/12/17 00:13:43.569207 [info] tls: using default ciphers

thank243, 2022-12-17:
编辑 `/etc/adguardhome.yaml`
修改如下配置
```
clients: 
  runtime_sources: 
    hosts: false
```
原因是os库升级了，默认不带 `/` 开头。本来是 `/etc/hosts`, 结果变成了 `etc/hosts` 系统无法找到文件导致panic

ac1595, 2022-12-29:
> 编辑 `/etc/adguardhome.yaml` 修改如下配置
> 
> ```
> clients: 
>   runtime_sources: 
>     hosts: false
> ```
> 
> 原因是os库升级了，默认不带 `/` 开头。本来是 `/etc/hosts`, 结果变成了 `etc/hosts` 系统无法找到文件导致panic。或者直接升级最新版本

这是高手！

minisind, 2023-11-22:
[image]
问题总是发生，这是什么原因啊？

thank243, 2023-11-22:
> [image] 问题总是发生，这是什么原因啊？

没有日志？

minisind, 2023-11-22:
MT7981的OPENWRT路由器，我是96年开始玩486电脑的老年人，做过一点A51,C51，一直在WIN平台转，今年才开始学习LINUX和OPENWRT。
刚刚重新定制固件，刷路由器了，玩ADBYBY，路由器数据清了。 感觉ADGH的LUCI不太友好，因为我连续4天左右弄这个ADGH，刷了很多次路由器，只成功了2次，并且在增加过滤规则时，卡死路由器无法工作了，只好不停刷机。 
相信ADGH的效果很好很强大，市场上的口碑很好，不过LUCI 和OPENWRT的适配也许可以更进步，有部分原因是我对LINUX不熟悉，还在不断学习中。

thank243, 2023-11-23:
> MT7981的OPENWRT路由器，我是96年开始玩486电脑的老年人，做过一点A51,C51，一直在WIN平台转，今年才开始学习LINUX和OPENWRT。 刚刚重新定制固件，刷路由器了，玩ADBYBY，路由器数据清了。 感觉ADGH的LUCI不太友好，因为我连续4天左右弄这个ADGH，刷了很多次路由器，只成功了2次，并且在增加过滤规则时，卡死路由器无法工作了，只好不停刷机。 相信ADGH的效果很好很强大，市场上的口碑很好，不过LUCI 和OPENWRT的适配也许可以更进步，有部分原因是我对LINUX不熟悉，还在不断学习中。

建议学习下这篇文章。
https://github.com/ryanhanwu/How-To-Ask-Questions-The-Smart-Way/blob/main/README-zh_CN.md

gita378, 2025-06-14:
istoreos可以在启动项中把‘停用’改为‘启用’解决

runner628, 2026-11-13:
感谢，终于解决